FreeCAD/FreeCAD #10729: [Problem] Missing API documentation for src/Mod/Part/App/GeomPlate

luzpaz, 2023-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Part/App/GeomPlate XML files show missing documentation about said GeomPlate functionality. 

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/GeomPlate/PointConstraintPy.xml#L78-L97

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/GeomPlate/CurveConstraintPy.xml#L78-L107

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/GeomPlate/CurveConstraintPy.xml#L117-L134

https://github.com/FreeCAD/FreeCAD/blob/4293c4a65b0e69975dcdbbf68c80d8019a4f850d/src/Mod/Part/App/GeomPlate/PointConstraintPy.xml#L78-L96

### Full version info

```shell
Current v0.22.x master
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

RohanMishra315, 2023-10-24:
hey @luzpaz can you provide the source , where its documentation located.

RohanMishra315, 2023-10-25:
Updated the api documentation @luzpaz

RohanMishra315, 2023-10-25:
Updated the docs @luzpaz

RohanMishra315, 2023-12-09:
@luzpaz  why this one is still open?

luzpaz, 2023-12-12:
Thanks @RohanMishra315   
JFYI, if you want to remotely automagically close a ticket from a PR checkout https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword#linking-a-pull-request-to-an-issue-using-a-keyword

closed 2023-12-12 by luzpaz