FreeCAD/FreeCAD #10733: [Problem] TechDraw: Expose SVG export edges

pieterhijma, 2023-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With two functions exposed from TechDraw, Python code and workbenches in particular would have access to generated SVG images based on Part HLR without having to duplicate code that is in the TechDraw module.

It is independent of but related to #10726. In order to test it with the Part HLR workbench it is necessary to have the changes from #10727.

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34337 (Git)
Build type: Unknown
Branch: master
Hash: 4293c4a65b0e69975dcdbbf68c80d8019a4f850d
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * osh-autodoc-workbench
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

An example of what would be nice to have is this [forum post](https://forum.freecad.org/viewtopic.php?p=699331#p699331).

I will issue a pull request shortly that allows the example in the forum post.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-03:
@pieterhijma is this ticket complete ?

pieterhijma, 2023-10-04:
Yes, by means of PR #10734.

closed 2023-10-04 by pieterhijma