FreeCAD/FreeCAD #10736: [Problem] exported STEP looks nothing like FCStd project

khimaros, 2023-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

two FCStd files with nearly identical feature trees produce very different STEP output when exported. a revolution feature ends up turning into a pocket during export.

i've created a simple reproduction case for this:

- open the attached FCStd file
- export the example-part part to STEP
- open the exported STEP and observe the odd pocket
- change the value of `Spreadsheet.revolution_x_offset` from `2.55mm` to `-2.55mm`
- export the same part again
- open the STEP and observe that it now looks correct

[example-part-bad.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/12662140/example-part-bad.FCStd.gz)

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.5, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * parts_library (Disabled)
  * sheetmetal 0.2.63
  * kicadStepUpMod 10.19.4
  * fasteners 0.4.61 (Disabled)
  * Assembly3 0.12.0
  * Manipulator 1.5.0
  * InventorLoader 1.4.0 (Disabled)
[/code]
```

### Subproject(s) affected?

File formats

### Anything else?

i've unzipped the FCStd files and compared the Document.xml between them. nothing really obvious jumped out there, so it seems like maybe there is an issue with the BREP?

because the shape is being inverted and turning into a hole instead of an extent, it feels almost a bit like an integer overflow or something of that sort.

i notice that when looking at the FCStd in the "bad" state, the revolution does not have all of its meridian lines, but it does after changing to the "good" state.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

khimaros, 2023-09-19:
perhaps of interest:

- exporting to STL works fine
- exporting to IGES results in completely empty geometry and report view says `07:59:57  'example-part' is not a shape, export will be ignored.`

perhaps recompute is silently failing?

luzpaz, 2023-09-19:
Please try to reproduce on v0.21.1 (latest stable)

khimaros, 2023-09-19:
@luzpaz confirmed it is still broken with the following release:

```
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at f670854)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

khimaros, 2023-09-20:
would it be helpful for me to include the two STEP files produced by the reproduction steps above?

luzpaz, 2023-09-20:
Yes please

Predko, 2023-10-20:
I have looked into this issue and have come to the conclusion that importing and exporting does not cause the error.
Error in Constraint2 = -1.25 mm in Sketch002.
Should be +1.25 mm.
I tried to repeat sketch002 with these restrictions, it turned out like this:
[image]

If you remove this constraint, then you can repeat the position of the circle only with a value of +1.25 mm.
Problem in sketch002

luzpaz, 2023-10-20:
Closing as user error

Predko, 2023-10-21:
@luzpaz ,
I'm sorry, but I was careless and made a mistake.
Problem still exists.