FreeCAD/FreeCAD #10739: [Problem] FreeCAD crashes when trying to apply constraint on external geometry line

Anthony-Gaudino, 2023-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashes when trying to **constrain horizontal distance**, **constrain distance**, **constrain vertical distance**, **constrain angle**  using external geometry.

Steps to reproduce:

1. Have some element like a cube.
2. Create a sketch and edit it.
3. Get one of the cube edges as external geometry.
4. Select the external geometry line and then click on one of the buttons for the constraints listed above.
5. FreeCAD crashes.

-----

I think FreeCAD should display the constraint as a reference constraint, for example, length as reference.

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.4
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.0
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

Similar to my previously reported crash on https://github.com/FreeCAD/FreeCAD/issues/9981

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-09-21:
I've failed to reproduce the bug on

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34369 (Git)
Build type: Debug
Branch: master
Hash: e0b814ed59c5f7758949b72291a69ea934a77482
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

and 

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git)
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

It seems to be MacOS-specific.

ollar, 2023-09-30:
same problem, macOS with m1 processor. Any ideas in finding solution?

luzpaz, 2023-10-11:
@chennes @PLChris could you test this on macOS ?

luzpaz, 2023-10-11:
@ollar full [about](https://wiki.freecad.org/About) info please

JohnOCFII, 2023-10-11:
I am able to reproduce this on my M1 MacBook Pro running Ventura 13.6

```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

Initial data from CrashDump:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [11874]
Path:                  /Applications/Other Apps/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [11868]
User ID:               501

Date/Time:             2023-10-11 15:21:03.1773 -0500
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       C550D563-E7FD-4681-81DC-2F61A49E5720

Time Awake Since Boot: 690000 seconds
Time Since Wake:       4518 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x000000010038127c
Exception Codes:       0x0000000000000002, 0x000000010038127c

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [11874]

VM Region Info: 0x10038127c is in 0x100338000-0x1003f4000;  bytes after start: 299644  bytes before end: 470403
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      shared memory               100334000-100338000    [   16K] r--/r-- SM=SHM  
--->  __TEXT                      100338000-1003f4000    [  752K] r-x/rwx SM=COW  ...c++.1.0.dylib
      __DATA_CONST                1003f4000-1003fc000    [   32K] r--/rw- SM=COW  ...c++.1.0.dylib

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libc++.1.0.dylib              	       0x10038127c std::__1::moneypunct<char, false>::do_curr_symbol() const + 0
1   libc++.1.dylib                	       0x1a71b1a38 std::__1::basic_ostream<char, std::__1::char_traits<char>>::operator<<(unsigned long) + 404
2   SketcherGui.so                	       0x16ad8c1c8 void boost::io::detail::put<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&, boost::io::detail::format_item<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>::string_type&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>::internal_streambuf_t&, std::__1::locale*) + 596
3   SketcherGui.so                	       0x16ad8bea4 void boost::io::detail::distribute<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>&, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&) + 180
4   SketcherGui.so                	       0x16ad8bd88 boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>& boost::io::detail::feed_impl<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>&, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&) + 248
5   SketcherGui.so                	       0x16add19e8 void Gui::cmdAppObjectArgs<unsigned long, char const (&) [6]>(App::DocumentObject const*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, unsigned long&&, char const (&) [6]) + 196
6   SketcherGui.so                	       0x16add95dc CmdSketcherConstrainDistanceX::activated(int) + 2148
7   libFreeCADGui.dylib           	       0x1018116d0 Gui::Command::_invoke(int, bool) + 520
8   libFreeCADGui.dylib           	       0x10181148c Gui::Command::invoke(int, Gui::Command::TriggerSource) + 332
9   libQt5Core.5.15.8.dylib       	       0x103a09424 0x1038c4000 + 1332260
10  libQt5Core.5.15.8.dylib       	       0x1039ec71c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
11  libQt5Core.5.15.8.dylib       	       0x1039ec71c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
12  CoreFoundation                	       0x1a7352560 __CFRunLoopDoSource0 + 176
13  CoreFoundation                	       0x1a73522d0 __CFRunLoopDoSources0 + 244
14  CoreFoundation                	       0x1a7350ed8 __CFRunLoopRun + 828
15  CoreFoundation                	       0x1a7350448 CFRunLoopRunSpecific + 612
16  HIToolbox                     	       0x1b0ba2df0 RunCurrentEventLoopInMode + 292
17  HIToolbox                     	       0x1b0ba2a80 ReceiveNextEventCommon + 220
18  HIToolbox                     	       0x1b0ba2984 _BlockUntilNextEventMatchingListInModeWithFilter + 76
19  AppKit                        	       0x1aa57797c _DPSNextEvent + 636
20  AppKit                        	       0x1aa576b18 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
21  AppKit                        	       0x1aa56af7c -[NSApplication run] + 464
22  libqcocoa.dylib               	       0x105928a64 0x1058fc000 + 182884
23  freecad                       	       0x1001a0568 main + 5064
24  dyld                          	       0x1a6f1bf28 start + 2236
```

JohnOCFII, 2023-10-11:
I can also reproduce the crash using this past weekend's weekly build:

```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: Release
Branch: master
Hash: e9187412d9b3b372ee4c1d7e2fc7a0bf08a2e647
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [12039]
Path:                  /Applications/Other Apps/FreeCAD-34594.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [12032]
User ID:               501

Date/Time:             2023-10-11 15:33:39.9170 -0500
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       C550D563-E7FD-4681-81DC-2F61A49E5720

Time Awake Since Boot: 690000 seconds
Time Since Wake:       5274 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000000102eb927c
Exception Codes:       0x0000000000000002, 0x0000000102eb927c

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [12039]

VM Region Info: 0x102eb927c is in 0x102e70000-0x102f2c000;  bytes after start: 299644  bytes before end: 470403
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      __LINKEDIT                  102e60000-102e70000    [   64K] r--/rwx SM=COW  ....5.15.8.dylib
--->  __TEXT                      102e70000-102f2c000    [  752K] r-x/rwx SM=COW  ...c++.1.0.dylib
      __DATA_CONST                102f2c000-102f34000    [   32K] r--/rwx SM=COW  ...c++.1.0.dylib

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libc++.1.0.dylib              	       0x102eb927c std::__1::moneypunct<char, false>::do_curr_symbol() const + 0
1   libc++.1.dylib                	       0x1a71b1a38 std::__1::basic_ostream<char, std::__1::char_traits<char>>::operator<<(unsigned long) + 404
2   SketcherGui.so                	       0x2b6710358 void boost::io::detail::put<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&, boost::io::detail::format_item<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>::string_type&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>::internal_streambuf_t&, std::__1::locale*) + 596
3   SketcherGui.so                	       0x2b6710034 void boost::io::detail::distribute<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>&, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&) + 180
4   SketcherGui.so                	       0x2b670ff18 boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>& boost::io::detail::feed_impl<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>&, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&) + 248
5   SketcherGui.so                	       0x2b674cff0 void Gui::cmdAppObjectArgs<unsigned long, char const (&) [6]>(App::DocumentObject const*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, unsigned long&&, char const (&) [6]) + 196
6   SketcherGui.so                	       0x2b675b438 CmdSketcherConstrainDistanceX::activated(int) + 2112
7   libFreeCADGui.dylib           	       0x10437d948 Gui::Command::_invoke(int, bool) + 516
8   libFreeCADGui.dylib           	       0x10437d708 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 332
9   libQt5Core.5.15.8.dylib       	       0x10659d424 0x106458000 + 1332260
[... 4 lines cut ...]
14  CoreFoundation                	       0x1a7350ed8 __CFRunLoopRun + 828
15  CoreFoundation                	       0x1a7350448 CFRunLoopRunSpecific + 612
16  HIToolbox                     	       0x1b0ba2df0 RunCurrentEventLoopInMode + 292
17  HIToolbox                     	       0x1b0ba2a80 ReceiveNextEventCommon + 220
18  HIToolbox                     	       0x1b0ba2984 _BlockUntilNextEventMatchingListInModeWithFilter + 76
19  AppKit                        	       0x1aa57797c _DPSNextEvent + 636
20  AppKit                        	       0x1aa576b18 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
21  AppKit                        	       0x1aa56af7c -[NSApplication run] + 464
[... cut ...]
32  AppKit                        	       0x1aa57797c _DPSNextEvent + 636
33  AppKit                        	       0x1aa576b18 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
34  AppKit                        	       0x1aa56af7c -[NSApplication run] + 464
35  libqcocoa.dylib               	       0x108678a64 0x10864c000 + 182884
36  freecad                       	       0x102ce0568 main + 5064
37  dyld                          	       0x1a6f1bf28 start + 2236
```

luzpaz, 2023-10-15:
Can someone make sense of the crash dump ?

xtemp09, 2023-10-16:
Well, the crash dump says that this is a duplicate of #10034.

The crash reported here occurs when processing the line https://github.com/FreeCAD/FreeCAD/blob/71a9eca750e3ccb7be122c4a8748c66da7fb07da/src/Mod/Sketcher/Gui/Utils.cpp#L643-L647 with `boost::format` present in https://github.com/FreeCAD/FreeCAD/blob/71a9eca750e3ccb7be122c4a8748c66da7fb07da/src/Gui/CommandT.h#L379-L382 
The crash reported in #10034 takes place when processing a similar line, like abdullah pointed out in [his post](https://forum.freecad.org/viewtopic.php?p=655134#p655134).

----

So, the error occurs when processing a seemingly harmless text with `boost::format` on _MacOS_. In 2013 I copied some files and launched _Skype_. This was my only interaction with _MacOS_. If I install it onto a virtual machine, I won't even know what button to press.

iAmVaeVictus, 2024-02-09:
Still a bug. 0.21.2-33771. Quite bothersome. 
Uploaded another Crash Report for good measure. 

[code]
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
[FreeCad Apple Crash Report.txt](https://github.com/FreeCAD/FreeCAD/files/14226408/FreeCad.Apple.Crash.Report.txt)

bozzzzo, 2024-03-08:
Commented on the issue #9981 , this is still a thing

JohnOCFII, 2024-03-09:
I am also no longer seeing this issue.  I am now running macOS 14.3, and when I saw the bug before I was running macOS 13.6.x

```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

chrisb-github, 2024-03-09:
It still crashes, but it is a bit more complicated.

To reproduce the issue:
1. create a Part cube
2. create a sketch in XY-plane, edit it
3. **create an arbitrary line**; this is new, without this the crash doesn't occur!
4. select the upper horizontal edge for external geometry
[image]

5. select the external geometry

6. create a horizontal distance constraint.

chrisb-github, 2024-03-09:
Without the additional step from my previous comment it reveals a regression, which finally may be the source of the problem:

1. create a Part cube
2. create a sketch in XY-plane, edit it
3. select the upper horizontal edge for external geometry
4. select the external geometry
5. create a horizontal distance constraint.

This creates a regular horizontal distance constraint (driving) ...
[image]
... while it used to create a driven one (blue)

xtemp09, 2024-03-09:
@chrisb-github, could you provide _About_ info and the backtrace?

JohnOCFII, 2024-03-09:
> It still crashes, but it is a bit more complicated.

I'm trying to replicate your crash, but it isn't happening for me:

```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```
[image]

[CrashAttempt.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14547750/CrashAttempt.FCStd.zip)

bozzzzo, 2024-03-10:
I've updated the OS, and I cannot reproduce the issue anymore

```
OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: C/Default (C) ]
Installed mods: 
  * ThreadProfile 1.89.0
  * ProDarkThemePreferencePack 1.0.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.4.72
  * lattice2 1.0.0
  * ExtremeProDark 2.9.0
  * Curves 0.6.15
```

maxwxyz, 2024-03-24:
@Anthony-Gaudino is this still relevant on current OS & FreeCAD?

Anthony-Gaudino, 2024-03-26:
> @Anthony-Gaudino is this still relevant on current OS & FreeCAD?

Currently I can't reproduce the problem anymore.
I have updated my OS, FreeCAD and Python, since I first created this issue.

Current version info:
```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.22
  * ThreadProfile 1.89.0
  * Launcher
  * Nodes 0.1.36
  * InventorLoader 1.4.0
  * woodworking 0.21.2.33771
  * Render 2024.1.23
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * nurbs
  * A2plus 0.4.64
  * Manipulator 1.5.7
  * fasteners 0.5.15
  * lattice2 1.0.0
  * CurvedShapes 1.0.8
  * sheetmetal 0.4.7
  * toSketch 1.0.1
  * Curves 0.6.29
  * Assembly4 0.50.12
  * Assembly3 0.12.2
```

closed 2024-03-26 by maxwxyz