FreeCAD/FreeCAD #10745: [Assembly] Card 1 : JCS are ugly

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 : 
The created JCS are fairly ugly.
@qewer33 the mockup you posted seems good to me. Do you want to attempt to implement it in coin or do you want to jump in ?

### Full version info

```shell
0.22 after https://github.com/FreeCAD/FreeCAD/pull/10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2023-09-20:
> Following the introduction of #10611 : The created JCS are fairly ugly. @qewer33 the mockup you posted seems good to me. Do you want to attempt to implement it in coin or do you want me to jump in ?

I can give it a shot but I need some pointers. What classes/files do I need to modify, or do I need to create a brand new class? Python or C++?

Also here are new mockups (with a circle alternative) with the plane on the center:

[image]

PaddleStroke, 2023-09-20:
It's python.
The coin code is in the JointObject.py ViewProviderJoint class

attach() is the function that creates the coin objects. and assign it to the display mode. In this function you can see : 
`self.switch_JCS1 = self.JCS_sep(obj, self.transform1)`
This is one joint coordinate system. So you can see its made in JCS_sep function
This function creates JCS = coin.SoAnnotation() this is basically the JCS coin object. You can see that the axis get added to this.
The axis themselves are create in the line_sep function.

The line thickness is a property, so it is actually defined in __init__.
The line lengths are applied by the self.axisScale. So its determined in get_JCS_size. Basically if you want to make them shorter you need to change this line : camera.height.getValue() / 10.
replace 10 by 20 and length will halve.
For the rectangle, you will need to create a soFaceSet (I seem to recall that's the name) And you will want to create it in a similar fashion as the lines are created.

howie-j, 2023-09-20:
The circle one looks nice :100: 
It will also help visually to make sure the JCS is placed correctly, e.g in the center of a hole.

PaddleStroke, 2023-09-20:
Alternate idea. Dunno what its worth.
[image]

qewer33, 2023-09-20:
Got something working! I'll also attempt to implement the view scaling for JCS's (https://github.com/FreeCAD/FreeCAD/issues/10746) and try to make it a circle somehow (is there like a helper function to calculate vertices from a circle w/ radius or something for SoFaceSet or do I have to write one myself? or use like NURBS node perhaps? or maybe even a really thin cylinder, not sure) 

[image]

PaddleStroke, 2023-09-21:
Great that's nice !
You can perhaps ask chatgpt he might have some ideals how to draw a circle.
Awesome if you can pick up the scaling as well !

commit 49a9f5b0 (2023-09-23): Assembly: Improve JCS appearance and implement autoscale

Closes #10745, closes #10746
closed 2023-09-23 by chennes