FreeCAD/FreeCAD #10747: [Assembly] Card 3 : JCS visibility should toggle when not in edit

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

When you finish to create a joint and close the taskbox, the JCS (joint coordinate systems) are still visible, which clutter the view.

The wanted behavior is : 
- JCS become hidden when their edition is left.
- JCS become visible when you select the joint in the tree view. And hides when the joint is not selected anymore.

### Full version info

```shell
0.22 after #10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-09-20:
I can give this one a go.

PaddleStroke, 2023-09-20:
Great ! 
Some indicators : 
The coin objects can be hidden by changing the SoSwitch nodes as it is done in showPreviewJCS.
I think the same can probably be used.

```
    def showPreviewJCS(self, visible, placement=None):
        if visible:
            self.switch_JCS_preview.whichChild = coin.SO_SWITCH_ALL
            self.set_JCS_placement(self.transform3, placement)
        else:
            self.switch_JCS_preview.whichChild = coin.SO_SWITCH_NONE
```
Though perhaps there is a better way that would hide / show the whole object.

sliptonic, 2023-09-21:
> I can give this one a go.

Very nice!  Please assign yourself to the issue and move it into the 'in progress' section of the project

PaddleStroke, 2023-09-21:
This is what I expected. For the second I'm not sure how this is done.
Basically you need to catch the event that the joint pythonFeature is selected in the tree from within the joint object to toggle visibility of its elements. Not sure how this can be done.

There is perhaps a function that gets called when a FeaturePython is selected similar to onChanged?

howie-j, 2023-09-25:
To solve the problem _JCS become visible when you select the joint in the tree view and hides when the joint is not selected anymore_ maybe we could write an observer class to handle tree selection events. This must then have callback methods for both `selected `and `unselected`.

PaddleStroke, 2023-10-06:
@howie-j Any update on that front ?

PaddleStroke, 2023-10-20:
@howie-j bump

howie-j, 2023-10-20:
Sorry for the delay. I have some free time this weekend so i plan on giving it a second go. However i'm not very experienced with gui programming, so i cant guarantee any results 😀

howie-j, 2023-10-22:
Made some progress on card 4, which i believe is a prerequisite for this issue.

@PaddleStroke I see that https://github.com/FreeCAD/FreeCAD/pull/10764 has done some refactoring, such as merging JointObject.py and CommandCreateJoint.py, and also solved Card 5. How far away do you think #10764 is from merging?

Just wondering if i should work on current master, or on the changes made in 10764..

PaddleStroke, 2023-10-23:
Thanks for having a look at my PR while doing yours. Indeed I changed some things. Such as moving the create joint task class to jointObject to avoid a dependency loop.
#10764 is a bit far from merging I think because it needs the ondsel solver to be made available first. Also its pretty big so review time is probably going to take a while.
So it depends if your changes are not much, then you can build on master and PR. And I'll rebase. If you need many changes, then perhaps its best to build on my branch and I will cherry pick your commits to add them to my PR perhaps?

closed 2024-02-14 by PaddleStroke