FreeCAD/FreeCAD #10748: [Assembly] When a joint is selected in the tree, its element should highlight

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

When a joint is not being edited, if you select it, the selected elements (face, edge or vertex) should be highlighted.

### Full version info

```shell
0.22 after #10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-09-23:
Are we sure we want the _element_ to be highlighted, not just the JCS?

PaddleStroke, 2023-09-24:
This is a good question. Perhaps it is not really helpful afterall. What really matter is the JCS I guess.

howie-j, 2023-09-24:
To highlight the selected JCS it's enough to set the property `"OnTopWhenSelected" = "Enabled"`, see this commit: https://github.com/howie-j/freecad/commit/940ed58412e9ba1845e8af795f418e2f10c8c74c

If that is sufficient, i can add it to my card 3 pr (not done yet)

PaddleStroke, 2023-10-20:
@howie-j It is not working on my side. The joint is hidden after accepting it. But selecting it does not show it nor does anything. Only toggling the visibility of the joint object shows the JCS.
Am I missing something?

howie-j, 2023-10-20:
Correct, it only highlights the selected joint when it's visible and selected in the tree view. https://github.com/FreeCAD/FreeCAD/issues/10747 must be solved first to actually make the joint visible when selected..

howie-j, 2023-10-22:
[card_3_4.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/6a525ae0-c678-47f0-af64-b0c9d44cbee0)

Some progress update:
Joints are now selectable in the 3d view and tree view, and are highlighed when selected. I wanted proper selection and highlighting to work before trying to solve card 3. A pr should be ready soon.

tobiasfalk, 2023-10-25:
> Are we sure we want the _element_ to be highlighted, not just the JCS?

I agree, but a way to also highlight the elements and not only the joints. 
Maybe wit a option kn the dropdownmenue(right click) of the joint.

pierreporte, 2023-11-07:
I don’t know if it’s better to write this here or on the PR. After seeing what @howie-j did this evening I thought of something that can be very useful when having parts very far apart: draw a line between the two JCSs. You could then follow the line and avoid mistakes. Catia does this and it saves the day. Whn you have a series of screws to put in holes, you’ll end up putting more than one screw in some holes, leaving some other empty. There could be an option: all lines visible (for all constraints), only selected constraint line is visible, no line visible.

PaddleStroke, 2024-01-08:
Also for distance joints, the distance could be writen  along this line in the same style as sketcher dimensions

pierreporte, 2024-01-08:
The dimensions should also appear for angle constraints.

Would the lines and dimensions open the constrain task box when double-clicked?

closed 2024-02-14 by PaddleStroke