FreeCAD/FreeCAD #10749: [Assembly] Card 5 : Joint should be editable.

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

Currently joints cannot be edited. Double clicking them should open back the joint taskbox to enable its modification.

### Full version info

```shell
0.22 after 10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-09-25:
There might be some hints in this thead: [https://forum.freecad.org/viewtopic.php?p=160045&hilit=doubleclicked#p160045](https://forum.freecad.org/viewtopic.php?p=160045&hilit=doubleclicked#p160045)

Basically a doubleClicked() method in the featurePython ViewProvider class get called when you doubleclick the joint object in the tree view. Then you can launch the createJointTaskPanel (or create a new class editJointTaskPanel, have not studied what is easier)

PaddleStroke, 2023-09-25:
Great so it shouldn't be too hard to do. I guess we just need to launch the createJointTaskPanel and set the correct selection. And on accept/reject handle things differently.

PaddleStroke, 2023-10-04:
Thanks for this detailed usecase. Well for let's get the solving and basic functionalities to work first, then we'll see what needs to be done. One step at the time.

sliptonic, 2023-10-04:
> Is there an official product specification based on user input, as interpreted by developers, that we users can review and comment? Or, is the strategy to use Onshape as the model? If so, then we users can review that product's capabilities and rest assured of the direction you are taking. I am assuming that you are the lead developer.

product specifications and detailed requirements are typical for a 'waterfall' style of development.  That doesn't happen much anymore in professional development shops and never really did in open-source. 

If we were following a professional model, we would have somebody like a 'product owner' who is technical enough to understand the development process but also knowledgeable about the application requirements.  The product owner helps set the interim goals and build the sprint targets.  Open-source is even looser than that.  We rely on consensus and constant feedback from the power users to set targets.  FWIW, having 'product owners' is something the project should consider.  It would honestly help.

@PaddleStroke isn't the lead developer.  (Well, he IS from Ondsel's perspective)   His/Our contributions to the workbench are just like anyone else's.  That means code contributions as well as issues.  If you think we're focusing on the wrong things or missing something important, it's important to communicate that by adding issues to the Github project. Consensus is the way forward.

PaddleStroke, 2023-10-20:
This card is done in https://github.com/FreeCAD/FreeCAD/pull/10764

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes