FreeCAD/FreeCAD #10750: [Assembly] Card 6 : In Joint taskbox, deleting an element in the list should unselect it.

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

In Joint taskbox, pressing delete when an element in the list is selected should unselect it.
Currently the behavior is undefined. So Delete press event is handled by the current selection. So if you click Delete, it delete the selected element. Which is of course unwanted.

### Full version info

```shell
0.22 after 10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-09-20:
I gave this one a go yesterday and got stuck.

My approach was to install an event filter in TaskAssemblyCreateJoint __init__
`        self.form.featureList.installEventFilter(self)`
or directly on the form as well : 
`        self.form.installEventFilter(self)`

```
    def eventFilter(self, watched, event):
        print(event.type())
        
        if watched == self.form and event.type() == QtCore.QEvent.KeyPress:
            if event.key() == QtCore.Qt.Key_Delete:
                selected_indexes = self.form.featureList.selectedIndexes()

                for i in selected_indexes:
                    if i < len(self.current_selection) :
                        selection_dict = self.current_selection[i]
                        Gui.Selection.removeSelection(selection_dict["doc_name"],selection_dict["obj_name"],selection_dict["sub_name"])

                return True  # Consume the event
        return super().eventFilter(watched, event)
```
But the problem is that because the user currently has a selection (as we are using greedy selection), the event is consumed before we receive it, and it is used to delete the selection. Then an event QtCore.QEvent.ShortcutOverride is emited.

So on top of the unwanted behavior of deleting the selection, we do not receive the KeyPress event as it is consumed.

So after thinking about it, I wonder if we don't want to drop the greedy selection, and do as PartDesign Fillet tool is doing : manually highligthing manually (see ViewProviderDressUp::highlightReferences). Or alternatively, as A3 is doing, ie creating subshape binders.

PaddleStroke, 2024-02-14:
@wwmayer Could you please weight in on this issue? Do you know how to prevent the 3dview selection from grabing the delete event when the QListWidget has the focus so that I can delete the item instead?

maxwxyz, 2024-02-16:
additionally you should be able to add or remove references in the joint task dialog.

wwmayer, 2024-02-17:
Usually if you want to override a key event that otherwise is handled by another instance you override the event() (keyEvent() may work too but don't know for sure) method and check for the type QEvent::ShortcutOverride. Then you check if the shortcut is what you want to handle. If yes, then either handle it directly or accept() it so that Qt sends a QKeyEvent that you must handle in keyEvent().

PaddleStroke, 2024-02-26:
@wwmayer Thanks that solved it

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre