FreeCAD/FreeCAD #10751: [Assembly] Card 7 : JCS snapping not ideal

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

Currently you cannot put your JCS to the middle of a line. Or to a face center. Which would be useful.
Maybe other cases are also missing / not ideal.

### Full version info

```shell
0.22 after 10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-09-27:
Is there some information what kind of snapping targets will be implemented or are currently on the road map?
I think between two faces and at an intersection of two faces would be great:
https://help.autodesk.com/view/fusion360/ENU/?guid=GUID-448D0BF0-6CD8-4E2F-AF94-58BAB7F716EC

Is it possible to get the axis of a cylinder by clicking on the outer face? What about the axis of a rotational symmetric object with no defined regular surface (lateral)?
Is it possible to reference created datum planes, points and lines/axis in a body/part for positioning or creating joints or offsets in the assembly?

PaddleStroke, 2023-12-08:
I have added line middle / center of gravity of faces (which handles cylinder axis)

Regarding your other points @maxwxyz : 

> What about the axis of a rotational symmetric object with no defined regular surface (lateral)?

Can you please give an example? I am not sure to understand. Do you mean a shape like a cog with dents on both sides ? Where not circle/arc/cylinder can be selected ?

> Is it possible to reference created datum planes, points and lines/axis in a body/part for positioning or creating joints or offsets in the assembly?

No yet. I will investigate this idea. I think it would be good to at least enable selecting custom local coordinate system within a body. So that they can be used for custom placements.

maxwxyz, 2023-12-08:
> Can you please give an example? I am not sure to understand. Do you mean a shape like a cog with dents on both sides ? Where not circle/arc/cylinder can be selected ?

yes, like a vase or or some surfaced shape (think of a human bone)

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes