FreeCAD/FreeCAD #10752: [Assembly] Joint Taskbox should have button to reverse direction if JCS and rotate it

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

Now there is no way to reverse the direction of the JCS nor to rotate it.
There should be : 
- A flip button.
- A rotate spinbox where you can specify a rotation angle. Next to it a 'turn 90°' button for convenience.

### Full version info

```shell
0.22 after 10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2023-09-20:
I propose to also add XYZ offset parameters along with flip and rotate.

tobiasfalk, 2023-09-20:
> I propose to also add XYZ offset parameters along with flip and rotate.

Yes sounds good, but a joint specific offset sounds mor sensible. Like a "Space" offset for the partalell and slide joint, here the specific space between the two parts is defined. Or a "Radial" offset distance and "radial" offset angle, wich define the offset to the centerline of a revolution joint.
This would make the offset more sensible and understandable to what they are offsetting.

qewer33, 2023-09-21:
> Yes sounds good, but a joint specific offset sounds mor sensible. Like a "Space" offset for the partalell and slide joint, here the specific space between the two parts is defined. Or a "Radial" [...]

That's also a really good idea but LCS offset is still necessary imo, they server different purposes. LCS offset is more for fine tweaking the alignment positions meanwhile joint offsets would be the actual alignment.

That would also probably need to be implemented at the solver side, I'm not sure how the proposed solver works as of yet.

PaddleStroke, 2023-09-21:
Beside general settings, some joints are going to need some specific settings which are not implemented yet.
Such has : 
- distance for parallel joints.
- Limits for slider / cylindrical.
- Perhaps limit for revolute too.
I have not created cards for those yet but it will come later I guess.

qewer33, 2023-09-21:
> Beside general settings, some joints are going to need some specific settings which are not implemented yet. Such has :

I would probably be better to implement these after the solver is integrated, since they will directly depend on what the solver actually supports

maxwxyz, 2023-12-02:
a rotation offset in general, also for fixed parts wold be great. Currently there is only a distance offset in one direction.

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes