FreeCAD/FreeCAD #10753: [Assembly] Card 9 : Joints do nothing.

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

Joint are currently doing nothing.

So now comes the actual solving. We need a way to solve the assembly using the joints.

Proposal : Make an API to the open 'Ondsel Solver' and connect assembly wb to it.
The big question is how to do it. I was thinking that the solving of the assembly would happen in the recompute of the assembly object, the App::Part. So that would mean making a subclass of App::Part in assembly, called Assembly::Assembly I guess. And in this class define the recompute to solve the assembly from its joints.

### Full version info

```shell
0.22 after 10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ppemawm, 2023-09-20:
Not being familiar with GitHub, this may not be the best place to leave this comment, but I know not where else:

In order to do top-down assembly **design** we need the ability to assemble an **empty** body to an existing body at the desired mating feature with the proper constraint and be able to work in the body link within the PartDesign workbench.  I.e. model in the context of the assembly.  This is possible with the current add-on workbenches.

Is this in your long term planning or will the integrated assembly workbench be limited to bottom-up assembly from existing models?

PaddleStroke, 2023-09-20:
I don't see why the wb wouldn't allow top down. But we have to start somewhere. When we get basic bottom up working properly, then we can see what is required to support top down.

When you say 'an empty body to an existing body at the desired mating feature with the proper constraint' can you please detail what you mean?
An empty body has per definition no feature to select. So the only one that'd be available is the origin of the body.
So you want to be able to add an empty body such that its origin matches a selected feature of the assembly? And that a joint is created using the selected JCS and the body origin as the second JCS? I don't see why this wouldn't be possible. But as I said it may be a little early. Better start this when bottom up is working ok.

jpg87Fr, 2023-09-20:
I think what I'm doing should answer the question, and the comments: an empty body... is not empty. At least for Assembly 3 WB!
[conception_train F16_EnSol2.pdf](https://github.com/FreeCAD/FreeCAD/files/12675907/conception_train.F16_EnSol2.pdf)

ppemawm, 2023-09-20:
@PaddleStroke @jpg87Fr 

"So you want to be able to add an empty body such that its origin matches a selected feature of the assembly? 
And that a joint is created using the selected JCS and the body origin as the second JCS?"

Yes and Yes.  At least that is what Assembly4 allows.  An Assembly4 Body includes a default universal  LCS mating connector that can be used for assembly before any feature is added.  The first sketch is located at this LCS placement.  There are many examples on the Users Showcase briefly describing this process.

It seems that this is a fundamental requirement of a usable assembly workbench such that it must be planned right from the beginning even if it is only implemented at a much later date.

sliptonic, 2023-09-20:
> It seems that this is a fundamental requirement of a usable assembly workbench such that it must be planned right from the beginning even if it is only implemented at a much later date.

Absolutely it should be planned from the beginning.  In fact, please create appropriate issues to steer development.  For example,  an issue like "Problem: user can't add an empty body to an assembly"    or "Problem: user can't build constraints to empty body LCS" would be appropriate.

As you said, it may only be implemented later.  Since our goal is to get something minimal working for the 1.0 release, I believe we should focus on bottom-up workflow right now as long as nothing we do precludes top-down design later.

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes