FreeCAD/FreeCAD #10754: [Assembly] Card 10 : 'Fix part' missing

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the introduction of https://github.com/FreeCAD/FreeCAD/pull/10611 :

We need a way to 'ground' or 'fix' parts.
This should perhaps take the form of a Joint as well. But with only 1 object obviously and no JCS required.

The question is, where do we put this. Some softwares use the mouse right click. I guess we could use a a toolbar entry. Perhaps in the joints toolbar, at the begining. And perhaps with a separator between it and other joints.

@qewer33 we need one more icon here :) and we need it to not be too confusing with the 'joint fixed' icon.

### Full version info

```shell
0.22 after 10611
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tobiasfalk, 2023-09-20:
What exactly would be the difference to Fixed joint? Does Fixed joint not already fix a part in space?

PaddleStroke, 2023-09-20:
Fixed joint goal is to fix two parts together through a contact point.
Fix part would be fixing a part in space.

Maybe the name 'Fixed joint' is not ideal it is prone to confusion with 'fix part'. If we can find another name that would be better.

howie-j, 2023-09-20:
I guess this command does not need the task view either, since you only need to select the object, not a JCS.
If i understand the use correctly, it's only purpose will be to prevent dragging an object around in space (or fix objects non-parametrically in an assembly)?

PaddleStroke, 2023-09-20:
Indeed, no taskbox required.
The point is usually to fix the first object. So that the whole assembly has an anchor part.
In solidworks for example, they just fix the first part you add in the assembly. Then you can unfix / fix by right clicking.
In onshape I think it's also a right click if I recall correctly.

qewer33, 2023-09-20:
> Indeed, no taskbox required. The point is usually to fix the first object. So that the whole assembly has an anchor part. In solidworks for example, they just fix the first part you add in the [...]

Does this really need a toolbar icon then?

Also anyone got any ideas for an icon? The lock icon is already used for the Fix Joint, I was thinking of changing the Fix Joint icon to be more like two parts interlocked together and using the lock icon for this instead.

howie-j, 2023-09-20:
> Also anyone got any ideas for an icon? The lock icon is already used for the Fix Joint, I was thinking of changing the Fix Joint icon to be more like two parts interlocked together and using the lock icon for this instead.

Yes! I was thinking just that. The padlock joint screams "locked" or "fix/ground" to me, unlike a "fixed joint" which should be to parts attached together via joint connectors.

tobiasfalk, 2023-09-20:
> Also anyone got any ideas for an icon? The lock icon is already used for the Fix Joint, I was thinking of changing the Fix Joint icon to be more like two parts interlocked together and using the lock icon for this instead.
> 
> 

Why not a vertical cylinder with a ground mark on the bottom.
[The ground like here on the top](https://images.app.goo.gl/cBtiZveGUCbvphn46)

qewer33, 2023-09-20:
> Why not a vertical cylinder with a ground mark on the bottom. [The ground like here on the top](https://images.app.goo.gl/cBtiZveGUCbvphn46)

That would be hard to show in a 32px icon and wouldn't be that clear I feel like.

qewer33, 2023-09-20:
Here is a new Fix Joint icon:

[image]

And proposal fix object icon:

[image]

PaddleStroke, 2023-09-20:
Indeed the names are maybe going to be confusing. Though perhaps not if the tooltips are clear. I don't know.
'Glue joint' has been mentioned before. 'Fastened joint' could be one, but not so original.

kadet1090, 2023-09-20:
I personally really like ground for fixing part in space and glue for fixing one part to another.

tobiasfalk, 2023-09-20:
> I personally really like ground for fixing part in space...

Then Why not fix to ground for the Fix part

kadet1090, 2023-09-21:
> Then Why not fix to ground for the Fix part

Just personal preference - If "Fix" can mean both then in my opinion it is better to choose wording that is not ambiguous for both operations.

PaddleStroke, 2023-10-06:
@qewer33 did you start working on this already ? If no please post the new icons and I'll tackle this now because I need it to solve assembly.

PaddleStroke, 2023-10-06:
I'm done with this card. Only the icons are missing. If you can send them it would be appreciated !

qewer33, 2023-10-06:
> I'm done with this card. Only the icons are missing. If you can send them it would be appreciated !

Sorry, I've been busy lately. It would also be great if you can tackle https://github.com/FreeCAD/FreeCAD/issues/10785. It's some quick JCS fixes suggested by wwmayer, shouldn't take too long.

And here are the icons:

[Assembly_.zip](https://github.com/FreeCAD/FreeCAD/files/12834172/Assembly_.zip)

Another note, it would be great if you can also use that new Fixed Joint icon on the "Joints" group under the Assembly container (instead of the current default folder group icon).

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes