FreeCAD/FreeCAD #10759: [UI Problem] 'Make Link' does not enable tool -> selection workflow.

PaddleStroke, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

'Make Link' does not enable tool -> selection workflow.
If you start the tool without something selected, it will create an empty link.

It would be better if the tool opened a taskbox to enable user to select an object.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-21:
FYI: You can drop an object onto an empty Link. Similar to what you can do with a Std_Group or Std_Part. There is an issue if you call Undo after the drop though.

PaddleStroke, 2026-04-26:
Yes but it's not very important I think. It could be closed.

closed 2026-04-26 by Roy-043