FreeCAD/FreeCAD #10760: [Problem] Crash when adding PartDesign fillet

Anthony-Gaudino, 2023-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to add a fillet to an edge FreeCAD crashes.

Steps to reproduce:

1. Click on edge as shown in image (green edge):
[image]

2. Click on add fillet button
3. FreeCAD crashes

Minimal file: [crash.zip](https://github.com/FreeCAD/FreeCAD/files/12671484/crash.zip)

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.4
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.0
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-20:
Did you double check the rest of fillet tickets ?

oscilococcinum, 2023-09-20:
I was able to add 0.3 fillet on that body, using nightly version.
[image]

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34291 (Git)
Build type: Release
Branch: master
Hash: 71f1e6b09baa45812026832dfd74e59b5d4d1d35
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Curves 0.6.13
  * DesignSPHysics 0.6.3(24-09-2022)
  * Plot 2022.4.17
  * sheetmetal 0.2.63
[/code]

xtemp09, 2023-09-21:
On Windows, FreeCAD crashlessly reports, though, without doing the fillet:

```ruby
13:27:43  <Exception> Access violation
13:27:43  Fillet: Access violation
13:27:44  <Exception> Access violation
13:27:44  Fillet: Access violation
```

Here is the backtrace:

<details><summary>gdb backtrace</summary>

```c++
Thread 1 "CrBrowserMain" received signal SIGSEGV, Segmentation fault.
0x00007ff93e269187 in TKMath!?Multiplied@TopLoc_Location@@QEBA?AV1@AEBV1@@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKMath.dll
(gdb) bt
#0  0x00007ff93e269187 in TKMath!?Multiplied@TopLoc_Location@@QEBA?AV1@AEBV1@@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKMath.dll
#1  0x00007ff93d8b45ca in TKBRep!?Surface@BRep_Tool@@SAAEBV?$handle@VGeom_Surface@@@opencascade@@AEBVTopoDS_Face@@AEAVTopLoc_Location@@@Z () from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKBRep.dll
#2  0x00007ff93d8c1293 in TKBRep!?Initialize@BRepAdaptor_Surface@@QEAAXAEBVTopoDS_Face@@_N@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKBRep.dll
#3  0x00007ff93e3adbfc in TKFillet!?StartSol@ChFi3d_Builder@@AEBA_NAEBV?$handle@VChFiDS_Spine@@@opencascade@@AEAV?$handle@VBRepAdaptor_Surface@@@3@AEAVgp_Pnt2d@@AEAV?$handle@VBRepAdaptor_Curve2d@@@3@AEANAEBV?$handle@VChFiDS_SurfData@@@3@_NH13AEA_N777126AEBVTopoDS_Vertex@@@Z () from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKFillet.dll
#4  0x00007ff93e3a84dd in TKFillet!?PerformSetOfSurfOnElSpine@ChFi3d_Builder@@AEAAXAEBV?$handle@VChFiDS_ElSpine@@@opencascade@@AEAV?$handle@VChFiDS_Stripe@@@3@AEAV?$handle@VBRepTopAdaptor_TopolTool@@@3@2_N@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKFillet.dll
#5  0x00007ff93e3a2537 in TKFillet!?PerformSetOfKGen@ChFi3d_Builder@@IEAAXAEAV?$handle@VChFiDS_Stripe@@@opencascade@@_N@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKFillet.dll
#6  0x00007ff93e3a7552 in TKFillet!?PerformSetOfSurf@ChFi3d_Builder@@IEAAXAEAV?$handle@VChFiDS_Stripe@@@opencascade@@_N@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKFillet.dll
#7  0x00007ff93e37127f in TKFillet!?Compute@ChFi3d_Builder@@QEAAXXZ ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKFillet.dll
#8  0x00007ff93e4e1c78 in TKFillet!?Build@BRepFilletAPI_MakeFillet@@UEAAXAEBVMessage_ProgressRange@@@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\TKFillet.dll
#9  0x00007ff93e8e7dbf in _PartDesign!?execute@Fillet@PartDesign@@UEAAPEAVDocumentObjectExecReturn@App@@XZ ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\lib\_PartDesign.pyd
#10 0x00007ff94de2a03a in FreeCADApp!?recompute@DocumentObject@App@@MEAAPEAVDocumentObjectExecReturn@2@XZ ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\FreeCADApp.dll
#11 0x00007ff93c64989a in Part!?recompute@Feature@Part@@MEAAPEAVDocumentObjectExecReturn@App@@XZ ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\lib\Part.pyd
#12 0x00007ff94ddc25d9 in FreeCADApp!?_recomputeFeature@Document@App@@IEAAHPEAVDocumentObject@2@@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\FreeCADApp.dll
#13 0x00007ff94de10956 in FreeCADApp!?recompute@Document@App@@QEAAHAEBV?$vector@PEAVDocumentObject@App@@V?$allocator@PEAVDocum--Type <RET> for more, q to quit, c to continue without paging--c
entObject@App@@@std@@@std@@_NPEA_NH@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\FreeCADApp.dll
#14 0x00007ff94ded35c4 in FreeCADApp!?recompute@DocumentPy@App@@QEAAPEAU_object@@PEAU3@@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\FreeCADApp.dll
#15 0x00007ff94decc84c in FreeCADApp!?staticCallback_recompute@DocumentPy@App@@SAPEAU_object@@PEAU3@0@Z ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\FreeCADApp.dll
#16 0x00007ff94c6010e8 in python310!PyCFunction_GetFlags ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\python310.dll
#17 0x00007ff94c5bc388 in python310!_PyObject_MakeTpCall ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\python310.dll
#18 0x00007ff94c6c0ad2 in python310!PyEval_GetFuncDesc ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\python310.dll
#19 0x00007ff94c6bcdf1 in python310!_PyEval_EvalFrameDefault ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\python310.dll
#20 0x00007ff94c6bf1c4 in python310!_PyEval_EvalFrameDefault ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\python310.dll
#21 0x00007ff94c731d51 in python310!PyRun_FileExFlags ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\python310.dll
#22 0x00007ff94c731e38 in python310!PyRun_FileExFlags ()
   from E:\temp\FreeCAD_weekly-builds-34310-2023-09-17-conda-Windows-x86_64-py310\bin\python310.dll
#23 0x00007ff94c731959 in python310!PyRun_StringFlags ()
[... cut ...]
Process 3108 exited with status = 1 (0x00000001) 
```

</details>

<details><summary>Version</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git) AppImage
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

----

Looking at the stacktraces, one can see that this OCCT bug reminds #6625 and #8086, but not the same.

xtemp09, 2023-09-21:
Æsir will let the developers of debuginfod enter Valhǫll out of turn.
<details><summary>Here is the backtrace showing where the bug in OCCT is located.</summary>

```c++
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
TopLoc_Location::Multiplied (this=0x7fffffff5c08, Other=...) at ./src/TopLoc/TopLoc_Location.cxx:109
Download failed: Invalid argument.  Continuing without source file ./obj-x86_64-linux-gnu/src/TKMath/./src/TopLoc/TopLoc_Location.cxx.
109	./src/TopLoc/TopLoc_Location.cxx: No such file or directory.
#0  TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff5c08, Other=...) at ./src/TopLoc/TopLoc_Location.cxx:109
#1  0x00007fffd5d166c3 in TopLoc_Location::operator*(TopLoc_Location const&) const (Other=..., this=<optimized out>, this=<optimized out>, Other=<optimized out>) at ./src/TopLoc/TopLoc_Location.hxx:101
#2  BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) (F=<optimized out>, L=...) at ./src/BRep/BRep_Tool.cxx:88
#3  0x00007fffd5d2b1fb in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) (this=0x5555597e8880, F=..., Restriction=Restriction@entry=true) at ./src/BRepAdaptor/BRepAdaptor_Surface.cxx:75
#4  0x00007fffd5aebb2e in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_HCurve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_HSurface>&, opencascade::handle<BRepAdaptor_HCurve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const (this=0x7fffffffa480, Spine=..., HS=..., pons=..., HC=..., W=@0x7fffffff60c8: -20.677609014566638, SD=..., isfirst=true, ons=2, HSref=..., HCref=..., RecP=@0x7fffffff60a9: false, RecS=@0x7fffffff60a8: false, RecRst=@0x7fffffff60a7: false, c1obstacle=@0x7fffffff60a6: false, HSBis=..., PBis=..., decroch=false, Vref=...) at ./src/ChFi3d/ChFi3d_Builder_2.cxx:1078
#5  0x00007fffd5af0d7d in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_HElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) (this=0x7fffffffa480, HGuide=..., Stripe=..., It1=..., It2=..., Simul=false) at ./src/ChFi3d/ChFi3d_Builder_2.cxx:2080
#6  0x00007fffd5af8abc in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) (this=0x7fffffffa480, Stripe=..., Simul=<optimized out>) at ./src/ChFi3d/ChFi3d_Builder_2.cxx:2625
#7  0x00007fffd5afbf29 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) (this=0x7fffffffa480, Stripe=..., Simul=<optimized out>) at ./src/ChFi3d/ChFi3d_Builder_2.cxx:3074
#8  0x00007fffd5ac3d95 in ChFi3d_Builder::Compute() (this=0x7fffffffa480) at ./src/ChFi3d/ChFi3d_Builder.cxx:264
#9  0x00007fffd5c21e0b in BRepFilletAPI_MakeFillet::Build() (this=this@entry=0x7fffffffa430) at ./src/BRepFilletAPI/BRepFilletAPI_MakeFillet.cxx:533
#10 0x00007fffd673c10d in PartDesign::Fillet::execute() (this=0x55555836a7d0) at /home/xtemp09/Downloads/FreeCAD/src/Mod/PartDesign/App/FeatureFillet.cpp:114
#11 0x00007ffff609de9f in App::DocumentObject::recompute() (this=0x55555836a7d0) at /home/xtemp09/Downloads/FreeCAD/src/App/DocumentObject.cpp:161
#12 0x00007fff536871d0 in Part::Feature::recompute() (this=<optimized out>) at /home/xtemp09/Downloads/FreeCAD/src/Mod/Part/App/PartFeature.cpp:92
#13 0x00007ffff6032d18 in App::Document::_recomputeFeature(App::DocumentObject*) (this=this@entry=0x5555585045d0, Feat=0x55555836a7d0) at /home/xtemp09/Downloads/FreeCAD/src/App/Document.cpp:3112
#14 0x00007ffff603aac8 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) (this=this@entry=0x5555585045d0, objs=std::vector of length 0, capacity 0, force=<optimized out>, hasError=hasError@entry=0x0, options=options@entry=0) at /home/xtemp09/Downloads/FreeCAD/src/App/Document.cpp:2839
#15 0x00007ffff6100203 in App::DocumentPy::recompute(_object*) (this=this@entry=0x5555584c2540, args=<optimized out>) at /home/xtemp09/Downloads/FreeCAD/src/App/DocumentPyImp.cpp:579
#16 0x00007ffff6100465 in App::DocumentPy::staticCallback_recompute(_object*, _object*) (self=0x5555584c2548, args=<optimized out>) at /home/xtemp09/Downloads/FreeCAD/build/src/App/DocumentPy.cpp:1949
#17 0x00007ffff5328008 in cfunction_call (func=0x7fffd62af1f0, args=<optimized out>, kwargs=<optimized out>) at ../Objects/methodobject.c:552
#18 0x00007ffff52dfadc in _PyObject_MakeTpCall (tstate=0x5555555f7e00, callable=0x7fffd62af1f0, args=<optimized out>, nargs=0, keywords=0x0) at ../Objects/call.c:215
#19 0x00007ffff527ba1c in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=9223372036854775808, args=0x7fffd6432e40, callable=0x7fffd62af1f0, tstate=<optimized out>) at ../Include/cpython/abstract.h:112
#20 _PyObject_VectorcallTstate (kwnames=0x0, nargsf=9223372036854775808, args=0x7fffd6432e40, callable=0x7fffd62af1f0, tstate=<optimized out>) at ../Include/cpython/abstract.h:99
#21 PyObject_Vectorcall (kwnames=0x0, nargsf=9223372036854775808, args=0x7fffd6432e40, callable=0x7fffd62af1f0) at ../Include/cpython/abstract.h:123
#22 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, trace_info=0x7fffffffb180, tstate=0x5555555f7e00) at ../Python/ceval.c:5893
[... cut ...]
#72 0x0000555555569538 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /home/xtemp09/Downloads/FreeCAD/src/Main/MainGui.cpp:272
```

</details>

<details><summary>The version of FreeCAD</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34369 (Git)
Build type: Debug
Branch: master
Hash: e0b814ed59c5f7758949b72291a69ea934a77482
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

</details>

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.

MiqCG, 2024-08-27:
I can reproduce the crash, too, with the provided file in Linux with the last revision (38553).

@luzpaz , I suggest that you remove the "Status: Needs test on dev version" label and add the labels you consider necessary to classify the issue. Thanks

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * PieMenu 1.8.0

AIRCAP, 2024-10-17:
still crashes 1.1 latest dev

pieterhijma, 2025-04-17:
I can confirm the crash. Note that I use a very recent OCC.

<details>
<summary>Version info</summary>
<pre>
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.41242 (Git)
Build date: 2025/04/14 08:05:41
Build type: Release
Branch: main
Hash: d56d90204565eb810a7adbdd6be8ad334c3e7142
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk , OCC 8.0.0.rc1-7aa85582ad
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
</pre>
</details>

@oscilococcinum provided a nice hint. Indeed with enforcing the radius to be 0.3 the fillet succeeds. This probably means that the extruded tube is in the way of successfully creating a fillet. Instead of an error message, we get a crash.

If I try to mimic this with a similar file, I do get an error message:

[image]

pieterhijma, 2025-04-18:
I created a minimal example with which I can reproduce the crash: [minimal.zip](https://github.com/user-attachments/files/19812906/minimal.zip)

The model looks like this and has more or less the same measurements as the similar features of the model with which this issue was reported:

[image]

In the example above, you can see that I got an error message. There are two features that seem to be important to make it crash:

The fact that there are two curves here:

[image]

And this rim:

[image]

Removing any of those two features gives a normal error message.

pieterhijma, 2025-04-18:
For reference, the stacktrace that I have in FreeCAD:

```
#0  0x00007fff80c8d108 in opencascade::handle<TopLoc_SListNodeOfItemLocation>::handle (this=0x7fffffff6e78, theHandle=...) at /home/pieter/scratch/dist-occt-main/include/opencascade/Standard_Handle.hxx:73
#1  0x00007fff80c8b767 in TopLoc_SListOfItemLocation::TopLoc_SListOfItemLocation (this=0x7fffffff6e78, Other=...) at /home/pieter/scratch/dist-occt-main/include/opencascade/TopLoc_SListOfItemLocation.hxx:57
#2  0x00007fff80c8b891 in TopLoc_Location::TopLoc_Location (this=0x7fffffff6e78) at /home/pieter/scratch/dist-occt-main/include/opencascade/TopLoc_Location.hxx:35
#3  0x00007fff76ab4fca in TopLoc_Location::Multiplied (this=0x7fffffff70c8, Other=...) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKMath/TopLoc/TopLoc_Location.cxx:98
#4  0x00007fff7b8b0d42 in TopLoc_Location::operator* (this=0x7fffffff70c8, Other=...) at /home/pieter/scratch/dist-occt-main/include/opencascade/TopLoc_Location.hxx:97
#5  0x00007fff7b133cc6 in BRep_Tool::Surface (F=..., L=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRep/BRep_Tool.cxx:79
#6  0x00007fff7b14d4c6 in BRepAdaptor_Surface::Initialize (this=0x55555b1d1480, F=..., Restriction=true) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRepAdaptor/BRepAdaptor_Surface.cxx:75
#7  0x00007fff8054cf05 in ChFi3d_Builder::StartSol (this=0x7fffffff8d00, Spine=..., HS=..., pons=..., HC=..., W=@0x7fffffff7430: -20.677609014566638, SD=..., isfirst=true, ons=2, HSref=..., HCref=..., RecP=@0x7fffffff7382: false, RecS=@0x7fffffff7383: false, RecRst=@0x7fffffff7384: false, c1obstacle=@0x7fffffff7385: false, HSBis=..., PBis=..., decroch=false, Vref=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:1199
#8  0x00007fff80553417 in ChFi3d_Builder::PerformSetOfSurfOnElSpine (this=0x7fffffff8d00, HGuide=..., Stripe=..., It1=..., It2=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:2420
#9  0x00007fff80557ae3 in ChFi3d_Builder::PerformSetOfKGen (this=0x7fffffff8d00, Stripe=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:3215
#10 0x00007fff8055af4e in ChFi3d_Builder::PerformSetOfSurf (this=0x7fffffff8d00, Stripe=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:3748
#11 0x00007fff80511ca6 in ChFi3d_Builder::Compute (this=0x7fffffff8d00) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder.cxx:258
#12 0x00007fff806b4562 in BRepFilletAPI_MakeFillet::Build (this=0x7fffffff8cb0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/BRepFilletAPI/BRepFilletAPI_MakeFillet.cxx:375
#13 0x00007fff780c8dd8 in BRepBuilderAPI_MakeShape::Shape (this=0x7fffffff8cb0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKTopAlgo/BRepBuilderAPI/BRepBuilderAPI_MakeShape.cxx:35
#14 0x00007fff7bc6dc4b in Part::TopoShape::makeElementShape (this=0x7fffffff93e0, mkShape=..., shapes=std::vector of length 1, capacity 1 = {...}, op=0x7fff7bd43d33 "FLT") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4143
#15 0x00007fff7bc6db5e in Part::TopoShape::makeElementShape (this=0x7fffffff93e0, mkShape=..., source=..., op=0x7fff7bd43d33 "FLT") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4130
#16 0x00007fff7bc6c03b in Part::TopoShape::makeElementFillet (this=0x7fffffff93e0, shape=..., edges=std::vector of length 1, capacity 1 = {...}, radius1=1, radius2=1, op=0x7fff7bd43d33 "FLT") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4006
#17 0x00007fff6cd473ff in PartDesign::Fillet::execute (this=0x55555e7b19d0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/PartDesign/App/FeatureFillet.cpp:99
```

pieterhijma, 2025-04-18:
I've also created a Draw Test Harness with as much as possible the same geometry as the FreeCAD minimal example: [test.txt](https://github.com/user-attachments/files/19813069/test.txt)

The stacktrace from running is:

```
#0  0x00007ffff7f4ec24 in opencascade::handle<TopLoc_SListNodeOfItemLocation>::handle (this=0x7fffffff9a18, theHandle=...) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKernel/Standard/Standard_Handle.hxx:73
#1  0x00007ffff7f4d1d3 in TopLoc_SListOfItemLocation::TopLoc_SListOfItemLocation (this=0x7fffffff9a18, Other=...) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKMath/TopLoc/TopLoc_SListOfItemLocation.hxx:57
#2  0x00007ffff7f4db6d in TopLoc_Location::TopLoc_Location (this=0x7fffffff9a18) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKMath/TopLoc/TopLoc_Location.hxx:35
#3  0x00007ffff5eb4fca in TopLoc_Location::Multiplied (this=0x7fffffff9c68, Other=...) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKMath/TopLoc/TopLoc_Location.cxx:98
#4  0x00007ffff790dd02 in TopLoc_Location::operator* (this=0x7fffffff9c68, Other=...) at /home/pieter/current/occt-liaison/occt-main/src/FoundationClasses/TKMath/TopLoc/TopLoc_Location.hxx:97
#5  0x00007ffff7553cc6 in BRep_Tool::Surface (F=..., L=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRep/BRep_Tool.cxx:79
#6  0x00007ffff756d4c6 in BRepAdaptor_Surface::Initialize (this=0x555555d60020, F=..., Restriction=true) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRepAdaptor/BRepAdaptor_Surface.cxx:75
#7  0x00007ffff274cf05 in ChFi3d_Builder::StartSol (this=0x555555984490, Spine=..., HS=..., pons=..., HC=..., W=@0x7fffffff9fd0: 9.2153981901626807, SD=..., isfirst=false, ons=2, HSref=..., HCref=..., RecP=@0x7fffffff9f22: false, RecS=@0x7fffffff9f23: false, RecRst=@0x7fffffff9f24: false, c1obstacle=@0x7fffffff9f25: false, HSBis=..., PBis=..., decroch=false, Vref=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:1199
#8  0x00007ffff2753417 in ChFi3d_Builder::PerformSetOfSurfOnElSpine (this=0x555555984490, HGuide=..., Stripe=..., It1=..., It2=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:2420
#9  0x00007ffff2757ae3 in ChFi3d_Builder::PerformSetOfKGen (this=0x555555984490, Stripe=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:3215
#10 0x00007ffff275af4e in ChFi3d_Builder::PerformSetOfSurf (this=0x555555984490, Stripe=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder_2.cxx:3748
#11 0x00007ffff2711ca6 in ChFi3d_Builder::Compute (this=0x555555984490) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/ChFi3d/ChFi3d_Builder.cxx:258
#12 0x00007ffff28b4562 in BRepFilletAPI_MakeFillet::Build (this=0x555555984440) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKFillet/BRepFilletAPI/BRepFilletAPI_MakeFillet.cxx:375
#13 0x00007ffff3563856 in BLEND (di=..., narg=5, a=0x555555593960) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKTopTest/BRepTest/BRepTest_FilletCommands.cxx:193
```

As you can see, it is crashing in the same path.

I will report this to OCCT.

pieterhijma, 2025-05-17:
This has been solved in the OCCT git repository. Confirmation that it doesn't lead to a crash any longer:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.41720 (Git)
Build date: 2025/05/16 13:27:33
Build type: Unknown
Branch: ir
Hash: 5a7b73e560157fbcb13f3f0289b1d072b2dbd677
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc1-3e80fad177
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
```

https://github.com/user-attachments/assets/4a11fee9-8616-4cae-a2e1-14ee3ab0f69e

stefano2734, 2026-09-18:
most 26.3  dev version are maintained with occt 7.x. And the issue is for them open.
Should be signed with label 8.0.0.

luzpaz, 2026-09-21:
Fixed upstream v8.0.1  
Users will see fix when #32431 is implemented

closed 2026-09-21 by luzpaz