FreeCAD/FreeCAD #10768: [Problem] The most important constraints are now grouped

leoheck, 2023-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently, the constraints were grouped. This is bad, since for me, they are the most used things in the Sketch toolbar.
How do I restore the previous view with them spread in the panel instead of inside of a list? Honestly, this change is awful..

[image]

### Full version info

```shell
[code]
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34367 (Git) AppImage
Build type: Release
Branch: master
Hash: e36e16de59dece2d67caf478d25e3a6b825102c6
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * WebTools 1.0.0
  * Defeaturing 1.2.0
  * Curves 0.6.13
  * fasteners 0.4.63
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * sheetmetal 0.3.0
  * Assembly4 0.50.2
  * kicadStepUpMod 10.19.4
  * Curves.backup1692246817.490895 0.6.12 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2023-09-21:
These things should not be grouped, they are really important.
There are other useless (or less useful tools in this panel that can be grouped saving space)
[image]

howie-j, 2023-09-21:
See https://github.com/FreeCAD/FreeCAD/pull/9810

The `Dimension` constraint tool (shortcut D) replaces all the other constraints for most use cases. It's really good, try it out!
If you wish to have the tools separate as before, there is an option for that in sketcher settings:

[image]

kadet1090, 2023-09-21:
The weird thing here is that the icon for the tool is from `Constraint diameter`, which is totally confusing.

howie-j, 2023-09-21:
No, it's because the diameter constraint is the last used one. The group/dropdown remembers the last used constraint.

kadet1090, 2023-09-21:
I don't know if it is good in that case? Dimension tool can replace all other dimensioning constraints, so in my opinion it should always be visible. If someone wants to use exact tool it's not that the person would want to use it again and again, probably it was used for one specific reason.

howie-j, 2023-09-21:
I totally agree.

qewer33, 2023-09-21:
> I don't know if it is good in that case? Dimension tool can replace all other dimensioning constraints, so in my opinion it should always be visible. If someone wants to use exact tool (like diameter) it's not that the person would want to use it again and again, probably it was used for one specific reason.

I think this is a limitation of Qt's toolbar button groups. But I agree it should be changed to be like that if possible.

leoheck, 2023-09-21:
If there is now one tool that does the job of all of those, all of them should be removed then, right? haven't tried yet.

howie-j, 2023-09-21:
Many people are used to the old tools and prefer to keep using them. The dropdown group was the result of a long forum discussion. [Forum thread](https://forum.freecad.org/viewtopic.php?t=65521&start=190)

leoheck, 2023-09-21:
So they create a really good tool, but they choose to keep the crap piled to annoy the users? If the new is better remove the others, if the others are good yet keep them available not grouped as broken toys.

qewer33, 2023-09-21:
> So they create a really good tool, but they choose to keep the crap piled to annoy the users? If the new is better remove the others, if the others are good yet keep them available not grouped as broken toys.

I don't get what you're trying to say here. It was discussed by a large group of users and the general consensus was to make the Dimension tool the default and have the other tools in a dropdown (because there may still be places where you might want to use a specific tool). This is also how it's done in other CAD software, SolidWorks being an example.

If you don't like it, just change it... `Preferences > Sketcher`

leoheck, 2023-09-22:
Thanks folks.

closed 2023-09-22 by leoheck