FreeCAD/FreeCAD #10770: [Feature request] transparency toggle.

PaddleStroke, 2023-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be nice to have a transparency toggle, when activated it switch a part between 0 and 63 transparency (63 is a 'good' value).
This way it would enable user to toggle transparency easily, without having to right click, appearance ... Which is too long if someone just want to quickly check what is behind a part.

Forum discuss : https://forum.freecad.org/viewtopic.php?t=81275

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-09-26:
Is there a possibility to integrate this with the concept of a show/no show space (or group of objects with currently 63% or hidden objects [maybe identified with 0% transparency]). Then switch the visibility between all visible objects and all hidden objects.
https://forum.freecad.org/viewtopic.php?p=665961

Reference from CATIA: http://catiadoc.free.fr/online/intug_C2/intugrf0220.htm

adrianinsaval, 2023-10-14:
alternatively, the view toolbar could have a combobox with a transparency slider. Or it could be in the status bar.

maxwxyz, 2023-10-20:
Currently when selecting an object in the 3D view and hitting space bar - the last Part Design Feature is hidden, not the body or part. This makes it hard to see in the collapsed tree view, which items are visible or not. Changing visibility ob the body does not toggle the last feature of the body.

Can this be tackled in this approach to toggle the visibility of the whole body, not the last Part Design feature of the selected body?

adrianinsaval, 2023-10-20:
I would consider that a separate issue, if you like you can check the issue tracker if there is an existing report and create one if there isn't.

maxwxyz, 2023-10-21:
Thanks, I have created a separate issue for that: https://github.com/FreeCAD/FreeCAD/issues/11132

PR #10805 (2023-11-06): Std_Transparency : Toggle transparency of objects.
Add a tool to toggle transparency.
[image]

It works with single solids. But also with App::Parts containing links to multiple files.
Fixes #10770
Forum discuss : https://forum.freecad.org/viewtopic.php?t=81275
closed 2023-11-06 by chennes