FreeCAD/FreeCAD #10773: [Problem] Missing API documentation for `src/Mod/Part/App/BRepFeat/MakePrismPy`

luzpaz, 2023-09-22:
https://github.com/FreeCAD/FreeCAD/blob/fdc44d50efb3c0c3cb0101a81f2b95f8d29a1f41/src/Mod/Part/App/BRepFeat/MakePrismPy.xml#L45-L49

luzpaz, 2023-09-23:
@wwmayer would know since he added the code originally https://github.com/FreeCAD/FreeCAD/commit/e1b56f855187b77282b4a751ea553fda8644751b

wwmayer, 2023-09-23:
https://dev.opencascade.org/doc/refman/html/class_b_rep_feat___make_prism.html#details

luzpaz, 2023-09-23:
https://dev.opencascade.org/doc/refman/html/class_b_rep_feat___make_prism.html#a9c19610716e6efd5afa1fafbeb1f3d4a

Thank wmayer! Any idea why upstream split the `perform` function in to 3 separate entities or whatever and didn't give them separate explanations ?
[image]

FlachyJoe, 2023-09-23:
The last text belongs the 3 cases.

1. Make prism from the Pbase face Length long
2. Make the prism from the Pbase face to the Until face
3. Make the prism from the From face to the Until face

_**BRepFeat_MakePrism.hxx:**_
```cpp
 //! Initializes this algorithm for building prisms along surfaces.
  //! A face Pbase is selected in the shape Sbase
  //! to serve as the basis for the prism. The
  //! orientation of the prism will be defined by the vector Direction.
  //! Fuse offers a choice between:
  //! -   removing matter with a Boolean cut using the setting 0
  //! -   adding matter with Boolean fusion using the setting 1.
  //! The sketch face Skface serves to determine
  //! the type of operation. If it is inside the basis
  //! shape, a local operation such as glueing can be performed.
  Standard_EXPORT void Init (const TopoDS_Shape& Sbase, const TopoDS_Shape& Pbase, const TopoDS_Face& Skface, const gp_Dir& Direction, const Standard_Integer Fuse, const Standard_Boolean Modify);
  
  //! Indicates that the edge <E> will slide on the face
  //! <OnFace>. Raises ConstructionError if the  face does not belong to the
  //! basis shape, or the edge to the prismed shape.
  Standard_EXPORT void Add (const TopoDS_Edge& E, const TopoDS_Face& OnFace);

  Standard_EXPORT void Perform (const Standard_Real Length);
  
  Standard_EXPORT void Perform (const TopoDS_Shape& Until);
  
  //! Assigns one of the following semantics
  //! -   to a height Length
  //! -   to a face Until
  //! -   from a face From to a height Until.
  //! Reconstructs the feature topologically according to the semantic option chosen.
  Standard_EXPORT void Perform (const TopoDS_Shape& From, const TopoDS_Shape& Until);
  
  //! Realizes a semi-infinite prism, limited by the
  //! position of the prism base. All other faces extend infinitely.
  Standard_EXPORT void PerformUntilEnd();
  
  //! Realizes a semi-infinite prism, limited by the face Funtil.
  Standard_EXPORT void PerformFromEnd (const TopoDS_Shape& FUntil);
  
  //! Builds an infinite prism. The infinite descendants will not be kept in the result.
  Standard_EXPORT void PerformThruAll();
  
  //! Assigns both a limiting shape, Until from
  //! TopoDS_Shape, and a height, Length at which to stop generation of the prism feature.
  Standard_EXPORT void PerformUntilHeight (const TopoDS_Shape& Until, const Standard_Real Length);
```

luzpaz, 2023-10-17:
Thanks @RohanMishra315 ! Closing as complete.

closed 2023-10-17 by luzpaz