FreeCAD/FreeCAD #10775: [Problem] Unknown CMake command "qt_find_and_add_translation".

grautvornix, 2023-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to build FreeCAD (on Windows), cmake aborts and prints the error that it cannot find the CMake command "qt_find_and_add_translation".
Apparently this function is defined in CMake\FreeCAD_Helpers\SetupQt.cmake but it is not included when configuring
I searched but could not find this issue anywhere which is strange, so sorry if it was already registered

### Full version info

```shell
latest
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grautvornix, 2023-09-22:
forgot to paste in the full error:

<snip>
-- C:/github/FreeCAD/build/src/Base/RotationPy.cpp
-- C:/github/FreeCAD/build/src/Base/PlacementPy.cpp
-- C:/github/FreeCAD/build/src/Base/AxisPy.cpp
-- C:/github/FreeCAD/build/src/Base/UnitPy.cpp
-- C:/github/FreeCAD/build/src/Base/QuantityPy.cpp
-- C:/github/FreeCAD/build/src/Base/PrecisionPy.cpp
CMake Error at src/Base/CMakeLists.txt:156 (qt_find_and_add_translation):
  Unknown CMake command "qt_find_and_add_translation".
<snip>

chennes, 2023-09-22:
Is this built in a clean (new) directory? Or atop an existing build?

grautvornix, 2023-09-27:
It works after having corrected libpack setting, so  no bug

closed 2023-09-27 by grautvornix