FreeCAD/FreeCAD #10785: Leftovers from #10767

luzpaz, 2023-09-23:
_Originally posted by @wwmayer in https://github.com/FreeCAD/FreeCAD/issues/10767#issuecomment-1732266399_

Some remarks:
* In order to avoid possible problems with scripting you should consider moving the view provider code into a separate file
* In the view provider you shouldn't add a callback function that is linked to the camera of the active view because this is error prone. If you open a new 3D view from the View menu then auto-scaling doesn't work.

Instead you should use the class SoShapeScale:
```
from pivy import coin

cube = coin.SoCube()

type = coin.SoType.fromName("SoShapeScale")
scale = type.createInstance()
scale.setPart("shape", cube)

graph = Gui.ActiveDocument.ActiveView.getSceneGraph()
graph.addChild(scale)
scale.scaleFactor = 10
```

CC @qewer33

qewer33, 2023-09-23:
@wwmayer I can't find any documentation about `SoShapeScale`. Seems like it's a FreeCAD class and not an OpenInventor/Coin3D one? What differences does it have compared to `SoScale`? Does it auto scale shapes according to the view?

Also I took the view scaling approach from the transform dragger's source (it also works with the `SoFieldSensor` callback way), might make sense to also refactor that if `SoShapeScale` is better.

wwmayer, 2023-09-24:
> I can't find any documentation about SoShapeScale. Seems like it's a FreeCAD class and not an OpenInventor/Coin3D one?

Originally, it was developed by the makers of Coin3D but it's not an official OpenInventor class.

> Does it auto scale shapes according to the view?

Yes, that's what it does. It's all done internally and you don't have to bother with the zooming.

> Also I took the view scaling approach from the transform dragger's source (it also works with the SoFieldSensor callback way), might make sense to also refactor that if SoShapeScale is better.

It works as expected because when setting the view provider in edit mode it receives the currently active 3D viewer. Only in this viewer you see the draggers but not in any other 3D viewer that belongs to the document.

luzpaz, 2023-10-11:
bump @qewer33 for status

luzpaz, 2023-10-23:
@qewer33 ping

qewer33, 2023-10-26:
I was busy the last few weeks, will handle this this weekend hopefully

PaddleStroke, 2024-05-15:
@qewer33 was this solved ? It was concerning the JCS objects ?

maxwxyz, 2024-09-04:
any news or should I unassign you?

PaddleStroke, 2024-10-07:
So it seems to work however I need to set the SoShapeScale for each line / plane. I cannot put a single instance of it in JCS_sep like so : 

```
    def JCS_sep(self, soTransform):
        JCS = coin.SoAnnotation()
        JCS.addChild(soTransform)

        base_plane_sep = self.plane_sep(0.4, 15)
        X_axis_sep = self.line_sep([0.5, 0, 0], [1, 0, 0], self.x_axis_so_color)
        Y_axis_sep = self.line_sep([0, 0.5, 0], [0, 1, 0], self.y_axis_so_color)
        Z_axis_sep = self.line_sep([0, 0, 0], [0, 0, 1], self.z_axis_so_color)

        JCS.addChild(base_plane_sep)
        JCS.addChild(X_axis_sep)
        JCS.addChild(Y_axis_sep)
        JCS.addChild(Z_axis_sep)
        
        # Doesn't work : 
        scale = coin.SoType.fromName("SoShapeScale").createInstance()
        scale.setPart("shape", JCS)
        scale.scaleFactor = 50

        switch_JCS = coin.SoSwitch()
        switch_JCS.addChild(scale)
        switch_JCS.whichChild = coin.SO_SWITCH_NONE

        return switch_JCS
```

If I add one instance on each axis/plane then it works : 

```
    def line_sep(self, startPoint, endPoint, soColor):
        line = coin.SoLineSet()
        line.numVertices.setValue(2)
        coords = coin.SoCoordinate3()
        coords.point.setValues(0, [startPoint, endPoint])

        axis_sep = coin.SoAnnotation()
        axis_sep.addChild(self.draw_style)
        axis_sep.addChild(soColor)
        axis_sep.addChild(coords)
        axis_sep.addChild(line)

        scale = coin.SoType.fromName("SoShapeScale").createInstance()
        scale.setPart("shape", axis_sep)
        scale.scaleFactor = 50

        return scale

    def plane_sep(self, size, num_vertices):
        ...
        face = coin.SoFaceSet()
        face.numVertices.setValue(num_vertices)

        transform = coin.SoTransform()
        transform.translation.setValue(0, 0, 0)

        face_sep = coin.SoAnnotation()
        face_sep.addChild(transform)
        face_sep.addChild(draw_style)
        face_sep.addChild(material)
        face_sep.addChild(coords)
        face_sep.addChild(face)

        
        scale = coin.SoType.fromName("SoShapeScale").createInstance()
        scale.setPart("shape", face_sep)
        scale.scaleFactor = 50

        return scale
```

closed 2024-10-14 by chennes