FreeCAD/FreeCAD #10786: [Problem] FEM naming fix part 2

FEA-eng, 2023-09-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After resolving #10134, there are still some names with "constraint" left where they shouldn't be present. I will try to clean this up. On this occasion, I will also consider removing "FEM" from some names where it's unnecessary.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34367 (Git)
Build type: Release
Branch: master
Hash: e36e16de59dece2d67caf478d25e3a6b825102c6
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-09-23:
"constraint" occurrences left to be removed:
- Constraint fixed
- Constraint displacement
- Constraint force
- Constraint pressure
- Constraint temperature

luzpaz, 2023-09-23:
Thanks for doing this good work @FEA-eng!

FEA-eng, 2023-09-23:
> Thanks for doing this good work @FEA-eng!

Yeah, I'm trying to improve the FEM workbench as much as possible but I will probably need help with this issue as well. I can change the names without "FEM_" for now but the latter will be much more tricky to replace.

FEA-eng, 2023-09-23:
Ok, I think that I will leave those functions/files with "FEM_" and take care of the first group for now.

FEA-eng, 2023-09-23:
Crowdin strings for the first group for our reference:
https://crowdin.com/translate/freecad/559/en-pl#6593161
https://crowdin.com/translate/freecad/559/en-pl#6593157
https://crowdin.com/translate/freecad/559/en-pl#6498312
https://crowdin.com/translate/freecad/559/en-pl#6593187
https://crowdin.com/translate/freecad/559/en-pl#6593203

kaktusus, 2023-09-23:
Suggestions for new labels ?

FEA-eng, 2023-09-23:
> Suggestions for new labels ?

Added :-)

PR #10799 (2023-09-25): FEM naming fix part 2 (removes remaining incorrect "constraint" usage)
Follow-up on #10519, resolves #10786 - cleaning the remaining names that shouldn't use "constraint"
closed 2023-09-25 by wwmayer