FreeCAD/FreeCAD #10793: document info popup not honouring system date locale

timsoftgit, 2023-09-24:
if you hover over a document the date information of the file is shown in US format (m/d/yyyy) instead of current locale (British) which is d/m/yyyy
see example which was of a file created 12 september 2023, not 9th December 2023

[image]

this affects windows 0.21.0 release version and also the 10th Sept 2023 weekly build windows version.

luzpaz, 2023-09-24:
Please edit the OP and add your full [About](https://wiki.freecad.org/About) info. TIA

yorikvanhavre, 2023-09-25:
I can confirm this...

In the code it's set correctly:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Start/StartPage/StartPage.py#L95

The problem is that it asks for FreeCAD's 'TimeFormat' preference value:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Start/StartPage/StartPage.py#L91
Which cannot be set anywhere in the Start preferences.

I'd see a couple of possibilities:

1. We add that setting in the start page preferences
2. We stop using that and rely on the system. But how to obtain a default time format string from the system in python? I can't seem to find the right way...

luzpaz, 2023-10-24:
@yorikvanhavre any idea who would be interested in tackling this ?
@qewer33  ?

yorikvanhavre, 2023-10-27:
I can do this, but would love some help from Python gurus out there for the best way to convert a timestamp obtained in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/StartPage/StartPage.py#L176 into a format preferred by the system...

luzpaz, 2023-10-27:
@0penBrain can you weigh-in? TIA

0penBrain, 2023-10-27:
> I can do this, but would love some help from Python gurus out there for the best way to convert a timestamp obtained in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/StartPage/StartPage.py#L176 into a format preferred by the system...

Normally just using `'%c'` as time format is enough.
You can test with something like
```
import time
time.strftime("%c")
```

If it doesn't work, please check what `locale.getlocale(locale.LC_TIME)` returns.

luzpaz, 2023-11-04:
@yorikvanhavre :point_up_2:

yorikvanhavre, 2023-11-10:
Okay using `%c` seems to do it... Thanks @0penBrain

commit 0e939872 (2023-11-13): Start: Fixed default time format - fixes #10793
closed 2023-11-13 by sliptonic