FreeCAD/FreeCAD #10801: [Problem] 'Formula Editor' dialog title isn't necessarily obvious for beginners that it is related to 'Expressions'

luzpaz, 2023-09-24:
_Originally posted by @atlast-50 in https://github.com/FreeCAD/FreeCAD/issues/10601#issuecomment-1732620066_

> Possibly also it might be worth considering renaming the window from "Formula Editor" to "Expression Editor"? I'm new to freecad, and cad software in general, but when I was trying to find the hotkey for the "Enter an expression..." button my path was check tooltip -> search through hotkey configuration -> google looking for an "Expression hotkey" because I didn't read the title of the window that pops up to notice that it's called "Formula Editor", and this gave no helpful results -> more random googling until I found it.

> Everywhere else that I can see just refers to expressions as expressions, if this window is the only one that refers to them as formulas it would make sense to rename that to be consistent.

luzpaz, 2023-09-24:
Possible proposals:   
* Rename to 'Expression Editor'
* Rename to 'Formula/Expression Editor'
* Keep 'Formula Editor' but rename any mention of 'expressions' in FreeCAD to 'formulas'

luzpaz, 2023-09-25:
@yorikvanhavre weighs-in in https://github.com/FreeCAD/FreeCAD/issues/10601#issuecomment-1733176976

luzpaz, 2023-11-04:
This was addressed in #10830, correct @atlast-50 ?

atlast-50, 2023-11-06:
Yes that addressed this

closed 2023-11-06 by luzpaz