FreeCAD/FreeCAD #10802: [Problem] Pocket_Shape doesn't extend corners and keeps disabling "enable extensions" again

michaelosthege, 2023-09-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the Path tool, I'm trying to cut a rectangular pocket with its corners extended by half the tool diameter.

I'm 95 % positive that I had this working several weeks ago, but now, using versions 0.20.2 or 0.21.1 it no longer works.

I can reproduce this in fresh files by the following steps:
1. Using Part Design create a rectangular cuboid, padded down by e.g. 10 mm
2. Draw another rectangle into the cuboid and create a 5 mm pocket
3. Create a Path object for the Body
4. Using "Pocket_Shape" select the bottom face of the pocket and plan an "Offset" path with default settings
5. Go to "Extensions" and enable "Extend Corners"
6. Apply 👉 nothing happens
7. OK
8. Edit the Pocket_Shape again and go to "Extensions" 👉 it disabled itself

[image]

-----------

Is this a bug, or is it me not remembering correctly how to extend corners?

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
[/code]
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-29:
@LarryWoestman can you weigh-in on this Path ticket ?

LarryWoestman, 2023-09-29:
I will take a look, but don't expect too much since all of this code is new to me.

I currently know enough to be dangerous about the postprocessors and am still trying to learn more about the rest of the Path code.

LarryWoestman, 2023-09-29:
I appear to have duplicated the behavior on:

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34463 +1 (Git)
Build type: Release
Branch: staging
Hash: dd6a6d1d6cb75a4eb491e5e7c6e5a30c9d5bb532
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

Now to figure out what is supposed to be happening based on the existing documentation and/or code.

LarryWoestman, 2023-09-30:
As far as I can tell, the Pocket_Shape operation insists on having a bottom face or a ring of faces around the pocket if the pocket is "through all".  It won't work with edges.

The Extension operation works only on edges and will extend a "corner" between two edges.  So we have a conflict between what the Pocket_Shape Operation will allow and what the Extension operation needs.

The Extension code automatically disables Extensions if it discovers that there are no Extensions defined.  That matches the problem description.

I couldn't figure out any way to get the Pocket_Shape operation to work with edges.  There is a check box that says "use the edges" but it didn't seem to do anything.

All of this should be verified by someone who understands the code better than I do :-)

LarryWoestman, 2023-09-30:
Correction:  the "use the edges" checkbox is actually called "Use Outline".  I don't think it is intended to help with this issue, but rather to ignore any "internal" features in the selected face such as islands or other pockets.

LarryWoestman, 2023-10-01:
I have tried to follow the code and got to the following point:

At the point where I check the "Enable Extensions" checkbox, the "createItemForBaseModule function in Path.Op.Gui.FeatureExtension.py is finds the edges that are part of the face for the pocket, but the edges fail the "isValid()" test, so no edges are returned.

I don't understand the code well enough to understand whether this is expected or an error, but it seems to be a problem at my current level of understanding.  I also can't figure out why the edges aren't valid.

I think this is as far as I can go.

michaelosthege, 2023-10-01:
@LarryWoestman thank you for investigating! I have zero clue about the FreeCAD codebase so far, but could you link the important lines?

LarryWoestman, 2023-10-02:
The FeatureExtension.py file is in github at [https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Path/Path/Op/Gui/FeatureExtension.py](url), and the most important lines are from 351 to 358.

I'm not sure how much that will help, since there is a **lot** of background information that you need to be able to make sense out of the code.  You might want to look at [https://wiki.freecad.org/Path_scripting](url) as one place to start.  Also the Power Users Hub [https://wiki.freecad.org/Power_users_hub](url) has pointers to many useful web pages.

luzpaz, 2023-11-11:
Thanks @LarryWoestman for the analysis.

Ping @mlampert (any time to take a look at this ticket?)

tcurdt, 2024-05-11:
I also just ran into this. Even in situations where it does not get disabled it's being completely ignored.

[image]
[image]

luzpaz, 2025-03-31:
@tcurdt can you still reproduce on v1.0.0 and 1.1dev ?

tcurdt, 2025-03-31:
I haven't tried in a while. Will need to give it another try, @luzpaz

closed 2026-09-22 by sliptonic