FreeCAD/FreeCAD #10803: compressed .wrl (.wrz) silently fails to decompress

evils, 2023-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

opening a .wrz (a gzip compressed .wrl file) (either a FreeCAD export or a manually compressed one)
opens, but shows no content

[i'm using this file for my tests](https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/0a94ebca5a5523db9f6474c262ec3c72ea0241bc/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x01_P2.54mm_Horizontal.wrl?inline=false) but i believe any .wrl could be use to reproduce this issue
(open .wrl, see model, select model, Export..., change extension to .wrz, Save, open newly created .wrz, see nothing)

when starting FreeCAD from the terminal, it emits the following:
```
FreeCAD 0.21.1, Libs: 0.21.1RUnknown
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

connect failed: No such file or directory
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Coin error in zlib glue(): Unable to load zlib DLL/shared object.
Coin warning in SoInput::setBuffer(): Buffer seems to be in gzip format, but zlib support is not available.
Coin read error: Erroneous character(s) after end of scene graph: "". This message will only be shown once for this file, but more errors might be present
	Occurred at line   1 in <memory>
```

no such error is shown in the `Report view` in the FreeCAD GUI

[the build log from Nix](https://hydra.nixos.org/build/236131465/log) shows that zlib (version 1.3) was found
though it is noted as `found version "1.3.#define ZLIB_VERSION "1.3""`

note that a zip (not gzip) compressed .step file (.stpZ) successfully opens in FreeCAD

### Full version info

```shell
[code]
OS: NixOS 23.11 (Tapir) (sway/sway)
Word size of FreeCAD: 64-bit
Version: 0.21.1.Unknown
Build type: Release
Python 3.10.12, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
[/code]

0.20.2 from nixpkgs also exhibits this behaviour
as does 0.21.1 (Libs: 0.21.1R33694 (Git)) on linux mint using flatpak
```

### Subproject(s) affected?

None

### Anything else?

i came across this in investigating a similar (maybe the same) issue in KiCad, where using compressed .wrl (.wrz) files results in no errors but no 3D models in the 3D viewer
(zip compressed .step files (.stpZ) are also handled correctly there)

i'm filing an issue here because this project is much more directly related to the problem (and also affected)
and FreeCAD's ability to export a .wrz makes it an easier place to reproduce the problem

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-09-25:
I cannot reproduce the problem on Windows and on Ubuntu.

<details><summary>Windows 10 build 19044</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git)
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

<details><summary>Ubuntu 22.04.3 LTS</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34369 (Git)
Build type: Debug
Branch: master
Hash: e0b814ed59c5f7758949b72291a69ea934a77482
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

</details>

<details><summary>Ubuntu 22.04.3 LTS and AppImage</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git) AppImage
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

Looks like packaging issue. @evils, could you reproduce it with AppImage?

luzpaz, 2023-09-25:
I can't reproduce on FreeCAD_weekly-builds-34387-2023-09-24-conda-Linux-x86_64-py310.AppImage

```ruby
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git) AppImage
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

evils, 2023-09-25:
i did try the appimage but that doesn't seem to run on `NixOS`
(we have our own `appimage-run` thing, it's not the most robust thing, and i'm on wayland, which probably doesn't help)

```
FreeCAD_0.21.1-Linux-x86_64.AppImage installed in /home/evils/.cache/appimage-run/9b8a991934e9d1edd7aa61b7ae938339e15b253ed612fe1f554abc1db9b6ee6f
FreeCAD 0.21.1, Libs: 0.21.1R33694 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
Addon Manager Warning: Could not import QtWebEngineWidgets -- README data will display as text-only
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/evils/.cache/appimage-run/9b8a991934e9d1edd7aa61b7ae938339e15b253ed612fe1f554abc1db9b6ee6f/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ImportError'>: libpci.so.3: cannot open shared object file: No such file or directory
libpci.so.3: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/evils/.cache/appimage-run/9b8a991934e9d1edd7aa61b7ae938339e15b253ed612fe1f554abc1db9b6ee6f/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ImportError'>: libpci.so.3: cannot open shared object file: No such file or directory
libpci.so.3: cannot open shared object file: No such file or directory
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
```

i don't know why my friend on linux mint didn't try the appimage, but mint's package failed to install with:
```
Some packages could not be installed. This may mean that you have requested an impossible situation or if you are using the unstable distribution that some required packages have not yet been created or been moved out of Incoming.
```

ah, after fixing `appimage-run` (adding `pciutils`) the appimage does run and does not have this issue
for completeness, still getting this:
```
(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
```

so i guess the problem is limited to Nixpkgs and flatpak
anyone know who does the flatpak so we can notify them?

closing this issue as the problem isn't here, feel free to reopen if someone here wants to keep investigating

luzpaz, 2023-09-25:
@evils thanks for being very on top of this issue. Please open a ticket at https://github.com/flathub/org.freecadweb.FreeCAD

closed 2023-09-25 by luzpaz

hfiguiere, 2024-01-18:
but what is the dependency that handle that file format?