FreeCAD/FreeCAD #10807: [UI Problem] Tasks panel separation, user doesn't get asked if they wish to keep the old tabulated layout

Syres916, 2023-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While I can fully understand that some/maybe even a majority of users would like to have the Tasks panel separated as it is now in 0.22dev master, it's not in the FreeCAD ethos IMHO to not give the user a choice on the first run after an upgrade.

For example, I was trying some fairly simple Techdraw work over the weekend and just had enough of the fact that I kept having to expand/move the panel around in order to work efficiently, I wasn't aware that the user could even move it over the Model panel to have it back to how it was previously.

As I'm regularly using clean config files for testing, I made a once run dialog box asking the user whether they want to retain the old Tabulated or the new Separated layout. https://github.com/Syres916/FreeCAD/commit/28894e9def6d2cffef7e479876221520adbcf490 This would be particularly of use to users with only 1024x768 resolution.

This maybe seen as unnecessary which is fine but at least I have a proposal alongside my gripe, sorry constructive feedback.

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34403 (Git)
Build type: Release
Branch: master
Hash: 37e477d3e7ed5af91eaa4e71ed2e12bc8a9c89ec
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.3
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * freecad.gears 1.0.0
  * Curves 0.6.13
  * Help 1.0.3
  * fasteners 0.4.56
  * sheetmetal 0.3.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-09-25:
I agree on the principle, users should not be confronted with an abrupt change of UI, but after the 0.21 release, many devs agreed we should take a more daring stance with the master branch, not try to keep it stable and not advertise it as "good for end users". In other words, master is where we should be allowed to break things, and not be afraid to break things.

Now, your point is good. A transition, or at least a choice or even just a warning and a way to get back, should be shown to the user.

mwganson, 2023-09-25:
My 2 cents: I prefer the combo box.

yorikvanhavre, 2023-09-26:
Related to #10681

commit 03987164 (2023-09-30): Gui: fixes #10807: By default keep the old tabulated layout of ComboView and TaskView
closed 2023-09-30 by 0penBrain