FreeCAD/FreeCAD #10810: [Problem] Scrolling works inconsistently when dragging and dropping objects

atlast-50, 2023-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When dragging an object in the side bar, for example, dragging a sketch into a body, if you don't move the mouse scrolling through the list of objects is unresponsive and results in no movement. Moving the mouse around while scrolling partially fixes this issue, but it's inconsistent.

Weirdly when my CPU was maxed out from other programs running this issue disappears, outside of things being a bit laggy as you would expect

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-25:
Any possibility you can show the phenomena with a screencast ?

atlast-50, 2023-09-25:
[image]
Throughout the whole gif here I am randomly scrolling up and down. When the mouse is stationary nothing happens despite scrolling. With mouse movement it scrolls but is inconsistent, for comparison here's the scrolling I'm doing without dragging anything
[image]

Roy-043, 2023-09-26:
This works for me:
When dragging an item hover over the label of the first or the last item in the tree. Don't drag the item beyond the top of the list.

closed 2025-02-14 by github-actions[bot]