FreeCAD/FreeCAD #10821: [Problem] Results of CalculiX FEM simulations are displayed using 2 separate objects

FEA-eng, 2023-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, there are two separate results objects used for CalculiX simulation results:
- CCX_Results
- Pipeline_CCX_Results

which is very confusing for new users. 

The first object:
- shows the results only when activated
- always shows contours on the mesh
- can't show stress and strain components
- doesn't show the color legend even though the contour plot is used (which is even more confusing and IMO rather pointless)
- can show min/max for the selected variables and uses various units which can be annoying (e.g. depending on the value, stress is shown in kPa, MPa, GPa, ...)
- supports user-defined output specified via equations which is nice but, again, there's no legend so only min/max values make sense
- can show the deformed mesh (there's a slider to change the deformation scale factor)

The second object:
- shows the results all the time
- supports different display modes (mesh/no mesh and so on)
- can show stress and strain components
- shows the color legend
- can be significantly expanded when ParaView-based filters are added to it

I think that at some point it would be necessary to unify those results objects. Since we need the possibility to use filters, basically the results pipeline would have to be expanded to include missing options that CCX_Results object has.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git)
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-02-18:
@maxwxyz Can you remove the Bug label? This one is a feature request.

maxwxyz, 2024-02-19:
edited all your mentions, thanks for pointing it out.

luzpaz, 2026-04-25:
@FEA-eng is this still relevant ?