FreeCAD/FreeCAD #10824: [Problem] FEM Solver job control and Run solver calculations icons are unintuitive

FEA-eng, 2023-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Current icons for "Solver job control" and "Run solver calculations" are IMO highly unintuitive:

[image]

I would suggest something simple and actually symbolizing solver controls/analysis submission. Maybe just a play arrow symbol for "Run solver calculations" and the same symbol but with a small gear icon for "Solver job control".

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git)
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-28:
@FEA-eng Can you please post a request in the UI/UX subforum and link to this ticket (+  vice-versa)

FEA-eng, 2023-09-29:
> @FEA-eng Can you please post a request in the UI/UX subforum and link to this ticket (+ vice-versa)

Done: https://forum.freecad.org/viewtopic.php?t=81586

PR #10885 (2023-10-02): FEM: Added more intuitive FEM solver icons
Resolves #10824 - the new icons (created by [bitacovir](https://forum.freecad.org/memberlist.php?mode=viewprofile&u=3136)) are much more intuitive than the old ones

[image]
closed 2023-10-02 by yorikvanhavre