FreeCAD/FreeCAD #10829: [Problem] Spreadsheet Workbench export does not have an associated filetype

gooberdawg, 2023-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a Spreadsheet object is created in a Project, the "Export Spreadsheet to CSV' function does not save the file with a filetype. The .CSV filetype must be manually added on the exported file.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.2
  * CommandPanel
  * fasteners 0.4.66
  * Manipulator 1.5.0
  * slic3r-tools
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Predko, 2023-09-27:
/src/Mod/Spreadsheet/Gui/Command.cpp file:
I think to export the file we need to replace line number 254

QString formatList = QObject::tr("All (*)");

this line

QString formatList = QObject::tr("All (*);;CSV(*.CSV *.csv)");

and add this lines after line number 266 ( if (isValid) { ):

if ( !fileName.endsWith(".CSV", Qt::CaseInsensitive)) {
	fileName.append(".CSV")
}

in this case, if the file extension is not entered, it will be added automatically

to import the file we need to replace line number 194

QString formatList = QObject::tr("All (*)");

this line

QString formatList = QObject::tr("All (*);;CSV(*.CSV *.csv)");

Unfortunately, I can't check.

chennes, 2023-09-27:
Thanks for looking into this, @Predko -- I think that code is probably more or less correct. Two things: first, on GitHub you can get code to be properly formatted by putting it in single backticks (for inline code, `like this`), or triple-backticks for code blocks
```
like this
```
Second, when you say you can't check, do you mean that you can't compile FreeCAD? We'd be happy to help you get set up if you want to do so. You can also fork the repo and make a personal branch with your fix on it and let the CI run on it to see if it at least compiles. Then ask here and someone who does compile FreeCAD can give it a test run.

Predko, 2023-09-27:
Thank you @chennes !
Okay, I'll try

luzpaz, 2023-09-28:
@Predko Click https://github.com/Predko/FreeCAD/compare/Predko-patch-%2310829 and open a PR. Make sure it's against FreeCAD/master

PR #10889 (2023-10-02): Spreadsheet: export CSV file extension by default
Fixes #10829

This pull request replaces #10853 as a rebased branch instead

Compiled and tested.
commit c9315ffc (2023-10-02): Spreadsheet: export CSV file extension by default

Fixes #10829
commit 2187c8dc (2023-11-11): Spreadsheet: export CSV file extension by default

Fixes #10829
closed 2023-10-02 by yorikvanhavre