FreeCAD/FreeCAD #10831: Boolean operations error in Mesh workbench

alexandrkamenev, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I get an error in Mesh workbench when doing any Boolean operation (union or subtraction).

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
OpenSCAD version 2021.01 is installed.
[/code]

I created two meshes in the Mesh workbench using the Mesh_BuildRegularSolid button: a cube and a cylinder. Then I do a logical operation to union (or difference) them. And I get an error.

12:50:40  Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "C:\Program Files\FreeCAD 0.21\Mod\OpenSCAD\OpenSCADUtils.py", line 529, in meshoptempfile
    result = callopenscadmeshstring('%s(){%s}' % (opname,meshimports))
  File "C:\Program Files\FreeCAD 0.21\Mod\OpenSCAD\OpenSCADUtils.py", line 493, in callopenscadmeshstring
    tmpfilename = callopenscadstring(scadstr, 'stl')
  File "C:\Program Files\FreeCAD 0.21\Mod\OpenSCAD\OpenSCADUtils.py", line 238, in callopenscadstring
    outputfilename = callopenscad(inputfilename, outputext=outputext,\
  File "C:\Program Files\FreeCAD 0.21\Mod\OpenSCAD\OpenSCADUtils.py", line 186, in callopenscad
    raise OpenSCADError('OpenSCAD executable unavailable')
<class 'OpenSCADUtils.OpenSCADError'>: 'OpenSCAD executable unavailable'
12:50:40  shutil.which('openscad') did not return a result. Using fallback.
```

### Subproject(s) affected?

OpenSCAD

### Anything else?

[Test.zip](https://github.com/FreeCAD/FreeCAD/files/12734133/Test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-09-27:
В верстаке _Mesh_, я нажал _Объединить_, и всё работает.

```cpp
14:44:34  Geometries in cache: 2
Geometry cache size in bytes: 94768
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 146000
Total rendering time: 0:00:00.155
   Top level object is a 3D object:
   Simple:        yes
   Vertices:      108
   Halfedges:     324
   Edges:         162
   Halffacets:    114
   Facets:         57
   Volumes:         2
```

<details><summary>Release</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```
</details>

<details><summary>Development version</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git)
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

----

Почему ошибкa из верстака _OpenSCAD_? Если дажe из него, там пишется, что исполняемый файл _OpenSCAD_ не найден.

```
<class 'OpenSCADUtils.OpenSCADError'>: 'OpenSCAD executable unavailable'
```

Не указан `openscad.exe` в настройках. Верстак _OpenSCAD_, затем, Правка → Настройки → OpenSCAD, Иполняемый файл OpenSCAD.

[image]

Roy-043, 2023-09-27:
Read the manual please.
https://wiki.freecad.org/Mesh_Union
https://wiki.freecad.org/Mesh_Intersection
https://wiki.freecad.org/Mesh_Difference

Roy-043, 2023-09-27:
Closing this...

closed 2023-09-27 by Roy-043