FreeCAD/FreeCAD #10837: [Problem] Status messages precisions of doubles is too high

PaddleStroke, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
It has been said on the forum before at several places.
The precision of double values is needlessly high in status messages, which reduce readability and serve no purpose.
I suggest reducing the number of decimals to 1 or 2 max so that it will be much more readable and will clutter less the view.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2023-09-27:
I think that we should simply make this configurable. Different workflows require different precision settings, It can also vary by unit. e.g. for mm `0.01` resolution would probably be enough for most people, but for inches it should probably be `0.001`

Syres916, 2023-09-27:
As someone who has to work in Imperial units, then 1/32" isn't an infrequent size which is 0.03125" so to use the global preference for decimals is the way forward.

zeeb1337, 2023-09-30:
This feels like a design choice to make sure the FreeCAD can be used on very deep levels in the future. But I would vote for an option to toggle this on and off, or since you can change how many decimals you want shown in the options, it should be related to that setting.

PaddleStroke, 2023-11-10:
@wwmayer I have found 3 duplicates of code setting the preselect message in the status bar.
One in Selection.cpp, one in SoFCUnifiedSelection.cpp and one in SoFCSelection.cpp.
```
    snprintf(buf,512,"Preselected: %s.%s.%s (%f %s, %f %s, %f %s)"
                    ,CurrentPreselection.pDocName
                    ,CurrentPreselection.pObjectName
                    ,CurrentPreselection.pSubName
                    ,pts[0].first, pts[0].second.c_str()
                    ,pts[1].first, pts[1].second.c_str()
                    ,pts[2].first, pts[2].second.c_str());
```
Could you let me know what is the difference between these 3 cases ? The difference of the role of each of these file is not very clear to me. Is there some doc on the matter or can you give a small explanation please?
Thanks

PaddleStroke, 2023-11-10:
Note, changing %f to %.2f reduce the number of decimals to 2.
`    snprintf(buf,512,"Preselected: %s.%s.%s (%.2f %s, %.2f %s, %.2f %s)"`
Now if we want to use system preference it is going to be more tricky. Probably something like 

```
int numberDecimals = 2; // This can be any integer
char formatString[20];

// Create the format string, e.g., "%.2f"
snprintf(formatString, sizeof(formatString), "%%.%df", numberDecimals);

// Now use the created format string in your final snprintf call
snprintf(buf, 512, "Preselected: %s.%s.%s (%s %s,%s %s,%s %s)"
                , CurrentPreselection.pDocName
                , CurrentPreselection.pObjectName
                , CurrentPreselection.pSubName
                , formatString, pts[0].first, pts[0].second.c_str()
                , formatString, pts[1].first, pts[1].second.c_str()
                , formatString, pts[2].first, pts[2].second.c_str());
```
But its not working yet.

wwmayer, 2023-11-11:
> Could you let me know what is the difference between these 3 cases ? The difference of the role of each of these file is not very clear to me.

* Selection is the central part of the selection mechanism
* SoFCUnifiedSelection is the root node of the scene graph
* SoFCSelection is the old way to handle selection on a per object basis.

wwmayer, 2023-11-11:
>     ParameterGrp::handle hGrp = Gui::WindowParameter::getDefaultParameter()->GetGroup("Units");
    int numberDecimals = hGrp->GetInt("Decimals", 2);

Don't use the parameter group here but instead use UnitsAPI::getDecimals.

Then for the decimals there should be an upper limit of e.g. 6 because in the preferences the number of decimals for quantities can be up to 16 and this would be by far too much for the text in the status bar.

> snprintf(formatString, sizeof(formatString), "%%.%df", numberDecimals);

I don't know how safe this is. I vote to move to QString or std::stringstream for a safe alternative.

And since almost the identical code is used in three different places the code should be refactored and e.g. a helper class or function should be created.

PR #11344 (2023-11-20): [Core] Status bar : reduce number of decimals
Fixes #10837

@wwmayer can you give your opinion on this ? Also if you have an idea where we could put a function to avoid this 3 time copy paste.

Thanks
closed 2023-11-20 by chennes