FreeCAD/FreeCAD #10839: [Problem] PartDesign Pad profile link window can't select elements in 3D view

Anthony-Gaudino, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a pad in PartDesign I can select existing faces from an object to pad.
If my model breaks due to topological naming problem, then I could go to the pad and click on the three dots `...` button to open a `Link` window. From there I could click on `clear` button to remove all selections and then click on the correct faces to pad.
[image]
[image]

After updating to 0.21.1, this seems to be broken. After selecting a face it will not add the face, so I need to select the feature on the list manually and then type in the faces I want, example `Face5,Face6` which is more time consuming as I need to check which name the face have.

It seems that other tools also have this bug, for example, `sheetmetal workbench bend`, so it's not specific to only PartDesign and Pad tool.

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.0
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-27:
Works OK here:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
```

luzpaz, 2023-09-29:
Maybe you can provide an example file?

Anthony-Gaudino, 2024-07-12:
> Maybe you can provide an example file?

This happens always, if I just create a PartDesign cube, add a fillet and try to change the fillet Base it happens.

I think this only happens on macOS.

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.

luzpaz, 2025-01-26:
@Anthony-Gaudino can you reproduce on 1.0 or 1.1dev ?    
Can you attach an example file for quick reproducibility ?