FreeCAD/FreeCAD #10840: [Problem] PartDesign Link window moves to behind FreeCAD window when clicking FreeCAD window

Anthony-Gaudino, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In PartDesign some features, like Pad, have a `Property -> Profile` property with a three dots `...` (one dot on 0.21.1).
Clicking on this button a `Link` window opens.
[image]
[image]

The issue with this window is that clicking anywhere on the main FreeCAD window will move the Link window to behind it, so I can't access it. To access the Link window I need to move the main FreeCAD window.

The Link window should always stay on top of the main FreeCAD window to avoid this problem.

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.0
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Anthony-Gaudino, 2023-09-27:
> Works OK here:
> 
> ```
> OS: Windows 8 build 9600
> Word size of FreeCAD: 64-bit
> Version: 0.21.1.33668 +26 (Git)
> Build type: Release
> Branch: (HEAD detached at 0.21.1)
> Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
> Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
> Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
> Installed mods:
> ```

Probably this is MacOS specific. This problem also existed on 0.20.2.

luzpaz, 2023-09-28:
@PLChris can you verify on macOS ?

luzpaz, 2023-11-15:
@JohnOCFII can you reproduce on latest macOS dev build ?

JohnOCFII, 2023-11-15:
Yes, problem still exists in current weekly build.  It took me a while to find how to access the `Link` window, as it is the single`.` on the right of the Profile name that is selected to access it.  Hard to see!  

If this is a "regular" feature, and not some under-the-hood developer debugging tool, I'd wish for a more visible access method (in addition to keeping the Links window frontmost in the stack of FreeCAD windows).

[image]

```
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35024 (Git)
Build type: Release
Branch: main
Hash: 4ff00ef99869e35295a5a17b8f486901504c655b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

luzpaz, 2023-11-16:
@adrianinsaval is this a UI tweek you could offer ? Not sure how it would be made just for macOS though...any idea ?

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.

Also, screen recording:

https://github.com/user-attachments/assets/95cd1cea-8179-4941-9a33-cd0b36ae5bef

FEA-eng, 2025-09-17:
@maxwxyz Can you check if it's still relevant on Mac?

closed 2026-04-08 by chennes