FreeCAD/FreeCAD #10841: [Feature] Provide a way to opt out from using OS dark theme

Anthony-Gaudino, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On MacOS FreeCAD 0.20.2 default theme would always use the light OS theme. On FreeCAD 0.21.1, it uses the dark theme if the OS is set to use a dark theme.

I would not like to use dark theme because the contrast is not good, specially some workbenches use dark icons which reduce the contrast even more.

Many software/webpages provide a way to opt out to use the system dark theme, so I suggest a feature in FreeCAD to opt out to use the system dark theme.

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.0
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

### Subproject(s) affected?

None

### Anything else?

Forum question: https://forum.freecad.org/viewtopic.php?p=706975

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-27:
Have you tried selecting a Light theme in the FreeCAD preferences?
Edit -> Preferences -> General -> General -> Theme

luzpaz, 2023-09-27:
Themes can be changed in the Preferences as Roy-043 points out. This is an opt-out. 

The dark themes could be improved for sure.  It would be great if you gave feedback in a separate ticket showing what you mean. 

Closing this ticket.

closed 2023-09-27 by luzpaz

Anthony-Gaudino, 2023-09-27:
@Roy-043, @luzpaz 

I know that I can change the theme in preferences, but the theme options available completely change the UI, The other themes besides the Classic theme are not good, they don't only look not as nice but it also makes it harder to use FreeCAD and have some visual glitches.
Please see bellow:

Classic theme (with OS light theme):
[image]

Light theme:
[image]

For these reasons I think it would be best to have an option to use the classic theme and opt out from using the dark theme.

-----

This is what Classic theme (with OS dark theme) looks like:
[image]

luzpaz, 2023-09-28:
Am I the only one confused by the use of the phrase 'opt out' in this context ?  
You can use the classic theme if you want, switch back to classic. End of story, right? What is it you're asking for ? What would you like the opt-out to look like?

If you want to help improve the UI (which we do welcome constructive feedback about), that's a different story. BTW, you're going to need to be more explicit in what you're pointing out when you post screenshots. You'll need to annotate the changes or else we can't read you mind and understand what comparison and differences you're pointing to. Also we do need more feedback from macOS users about the UI/UX. There is a subforum dedicated to UI/UX issues. Can you join the forum and post some of your suggestions ?