FreeCAD/FreeCAD #10843: [Problem] FreeCAD crash when clicking on invalid fillet edge in side panel

Anthony-Gaudino, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a part which contains a fillet that results in an invalid shape. This happened from refactoring features.
If I double click on the fillet, and right click or left click on the edge that is listed in the selected list FreeCAD crashes.
[image]

File: [crash.zip](https://github.com/FreeCAD/FreeCAD/files/12738070/crash.zip)

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * lattice2 1.0.0
  * sheetmetal 0.3.0
  * toSketch 1.0.1
  * Curves 0.6.13
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-09-28:
I cannot reproduce it on Linux and Windows. FreeCAD writes, instead of crashing:

```java
13:09:40  Unhandled unknown exception caught in GUIApplication::notify.
13:09:40  Unhandled unknown exception caught in GUIApplication::notify.
13:10:07  top_block_front_fillet: Failed to create fillet
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34438 (Git)
Build type: Debug
Branch: master
Hash: 247df9939108ebe6cf398868e59f49d965c3e381
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

----

On Windows:

```java
13:11:48  Unhandled unknown exception caught in GUIApplication::notify.
13:11:48  Unhandled unknown exception caught in GUIApplication::notify.
13:11:55  Recompute failed!
```

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.

maxwxyz, 2024-10-07:
No crash but warning on Windows with the stable version:
[image]
```
[code]
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
[/code]

Same on latest dev:
![grafik](https://github.com/user-attachments/assets/c7fde442-f5a6-426b-946b-87c74de151a9)

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```
```

wwmayer, 2024-10-13:
In the meantime we know that if an event handler or a slot function doesn't handle exceptions they go through Qt code. On macOS this usually leads to a crash while on Windows or Linux only a warning is printed.

PR #17227 (2024-10-13): PD: Fix crash when clicking on invalid edge in task panel
Fixes #10843
commit f9547e82 (2024-10-13): PD: Fix crash when clicking on invalid edge in task panel

Fixes #10843
commit 7c842eea (2024-10-13): PD: Fix crash when clicking on invalid edge in task panel

Fixes #10843
closed 2024-10-13 by chennes