FreeCAD/FreeCAD #10846: DXF file not being exported

lstremfelj, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

on macOS Big Sur 11.7.10 With 0.20.1 version it was possible to export the drawing to dxf without any issue. Directly from file > export and in drop down selecting the autodesk .dxf format

Then I installed the new version 0.21.1 (unsigned)  this export stoped working reporting error : no moule named ezdxf

I decided to delete new version reinstal older one i had previusly installed and the export to dxf and it stoped running here also. I get error : module "importDXF" has no attribute "export" 

Tried the older 0.20.1 and new version 0.21.1 on windovs 10 OS there are no issues with the export function. Any idea how to solve this¿

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.89.0
  * freecad.gears 1.0.0
  * A2plus 0.4.60n
  * fasteners 0.4.66
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.3.0
[/code]

and 
[code]
OS: macOS 11.7.10
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.89.0
  * freecad.gears 1.0.0
  * A2plus 0.4.60n
  * fasteners 0.4.66
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.3.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-27:
@chennes can you reproduce on macOS ?

yorikvanhavre, 2023-09-29:
ezdxf is not used anywhere in FreeCAD. This error likely comes from an add-on. You should identify which addon is responsible for this (error messages could help), and file a bug report against that addon.

From your addons list, I'd say the culprit is Alternate_OpenSCAD...

luzpaz, 2023-09-29:
> From your addons list, I'd say the culprit is Alternate_OpenSCAD...

Correct, https://github.com/search?q=repo%3AKeithSloan%2FOpenSCAD_Alt_Import%20ezdxf&type=code

CC @KeithSloan

closed 2023-09-29 by luzpaz

lstremfelj, 2023-09-29:
Thank you  both. That was the problem. After I deleted the addon it works - happy as a radio.
Is it possible this addon was loaded automaticly because i had @allow FreeCDA to automatically download and update the DXF libraries" ? 
Or this comes from somewhere else cause i do not remeber manualz installoing this addon.

luzpaz, 2023-09-29:
@lstremfelj can you open a ticket on https://github.com/KeithSloan/OpenSCAD_Alt_Import/issues and reference this issue? That will help you get to the bottom of this and help development for OpenSCAD_Alt_Import addon as well. TIA!

KeithSloan, 2023-09-29:
@lstremfelj Please could I trouble you to reinstall the Alternate importer after refreshing the Addon Manager cache.

1) Installing via the Addon Manager should cause a prompt for the installation of the python ezdxf library.
2) With the workbench installed you should get offered a choice of which exporter to use.

lstremfelj, 2023-09-29:
@KeithSloan 

Yes, when instaling from Addon Manager it promted to install EZdxf library first, after being installed it continued installing OpenSCAD alternate importer. After Frecad restart I tried exporting the drawing as at begining and got same error as on begining - module "importDXF" has no attribute "export"

KeithSloan, 2023-09-29:
Thanks for testing - Have updated the Alternate Workbench to try and avoid conflicts with FreeCAD import/exporter.

luzpaz, 2023-09-29:
Maybe the addon manager can have some code that checks 3rd party addons for name conflicts like this one? @davesrocketshop RocketWB had a thing some time ago where there were name conflicts.