FreeCAD/FreeCAD #10847: FreeCAD can open, but not save a file with a Junction in the path (Win10)

tbird1965, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Because Windows' One Drive has very long paths (including spaces - ugh), I use a Junction as a shortcut on my filesystem:
```shell
Directory of D:\
09/14/2022  08:22 AM   <JUNCTION>     documents [d:\OneDrive - My Company Name\Documents]
``` 
If I have a FreeCAD file at d:\documents\bug\panel.FCStd, I can navigate to the junction `d:\documents\bug\panel.FCStd` and click it to open it via file explorer without issue:
[image]

However, when I hit save, I get the following error message:
[image]

To save it, I have to chose "Yes" and then navigate to the actual path `d:\OneDrive - My Company Name\Documents\bug\panel.FCStd` to save it.

For testing, I created a directory symlink: `09/27/2023  11:32 AM    <SYMLINKD>     doc [d:\OneDrive - My Company Name\Documents]`.  FreeCAD will open/save using a directory symlink, but since Windows only lets admins create directory symlinks and symlinks redirect differently when viewing via a remote system, I'd prefer using a junction.

I also noticed that using File | Open with a junction displays the junction path `d:\documents\bug`, but if I use File | Open and click a directory symlink, I see the original `D:\One Drive - My Company Name\Documents` path in the navigation bar.  This does not occur in File Explorer, so somehow the open dialog is doing something different.

FreeCAD File | Open when navigating using junction (desired display):
[image]

FreeCAD File | Open when navigating using directory symlink (not desired display):
[image]

Windows File Explorer when navigating using directory symlink (desired display):
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-27:
Why do use `<Junction>` in the filename ? I'm still not clear on that.

tbird1965, 2023-09-27:
If you are referring to the directory of D drive - that is just how the command prompt tells you there is a junction defined between the word `documents` and `D:\OneDrive - My Long Company Name is Here\Documents`

If you are asking why I even use a junction, at least in the Windows world, a junction is a local directory mapping.  It is like a file hard link but for directories.  Without the junction, my paths would be always very long and include spaces like:
```shell
D:\OneDrive - My Long Company Name is Here\Documents\projects\customers\custA\project23\size_26_panel\datasheets
``` 
With the junction, I can have this (and never see `D:\OneDrive - My Long Company Name is Here` or the spaces in the filename:
```shell
D:\documents\projects\customers\custA\project23\size_26_panel\datasheets
``` 
As a "standard" user, I can set and remove junctions without rights elevation.  Symlinks require elevation.

PS: Microsoft defines the OneDrive path with company name and spaces on all corporate PCs - you have no chance to change it - unless you change your company name - but it will always come with spaces around the hyphen.

luzpaz, 2023-09-27:
Any relationship with #5592 and #7596

tbird1965, 2023-09-27:
Both https://github.com/FreeCAD/FreeCAD/issues/5592 and https://github.com/FreeCAD/FreeCAD/issues/7596 refer to symlinked files only and am not sure if that really applies to directories.  FreeCAD seems to be 98% working with directory symlinks/junctions.  I don't have any problems of a second file appearing as mentioned in those tasks you list.  I just cannot click File | Save when I have opened the file using a directory junction.  Additionally, if I switch to a directory symlink, then the file dialog box shows the real path and not the symlink version - so technically "works", but is not as desired.

I have found that FreeCAD shows the file project info correctly (shows junction for the path) when you view File | Project Info (it just fails on File | Save).

`D:/documents/bug/panel_size26.FCStd` is the path with the junction in use correctly, which is shown in the project info.

The actual path I have to navigate to so I can actually resave my file after changes is (i.e. can't use the junction):

`D:\OneDrive - My Company Name\Documents\bug\panel_size26.FCStd`:

[image]

xtemp09, 2023-12-02:
User can create symlinks without elevation of privileges on Windows 10. [^1] I use them instead of junctions.

Qt has functions to avoid this kind of trouble, like `QString QFileInfo::junctionTarget()` and `std::filesystem::path QFileInfo::filesystemJunctionTarget()`, however, they were introduced in Qt 6.2. [^2] Maybe, it's for the best to address this issue after the transition to Qt 6.

[^1]: https://blogs.windows.com/windowsdeveloper/2016/12/02/symlinks-windows-10/
[^2]: https://doc.qt.io/qt-6/qfileinfo.html#filesystemJunctionTarget

tbird1965, 2024-01-08:
@xtemp09 - The Windows blog is interesting, but standard mode of Windows 10 still requires elevation.  User must go into Developer mode.  It is unclear since I am using a corporate machine, a) if I physically can (appears I am blocked), and b) what side effects will result.  Windows 11 looks to have dropped the admin requirement for symlinks.

Wiredhome, 2024-01-29:
I have the same issue - I cannot save into a folder on a junction path. For me, as soon as I click 'Save', FreeCAD hangs. Even with a "FreeCAD.log" file being generated, no new entries appear from the moment I click Save.

My use of Junction is for slightly different reasons - all users have established a consistent structure: C:\Projects, and within that are individual projects. There's no junction in this example, but several PCs didn't have a large C drive, so c:\Projects was the junction to D:\Projects. Others would like the C:\Projects to map to a network path.

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * pcb 6.2023.1
[/code]

File | Project Information shows:
[code]
C:/Projects/SweeperConcept/Mechanical/Enclosure_Main.FCStd
[/code]
even as this is on D:/Projects/SweeperConcept/...

I have tracked most of the FreeCAD releases since 0.16, and I did not use to have this problem [as best I recall], so I guess this came along with an update to Qt - perhaps 0.19?

luzpaz, 2024-09-20:
> Qt has functions to avoid this kind of trouble, like QString QFileInfo::junctionTarget() and std::filesystem::path QFileInfo::filesystemJunctionTarget(), however, they were introduced in Qt 6.2. [...]

This needs an `ifdef` solution for those who are using FreeCAD with Qt6 libs. Anyone able to spearhead this ?

luzpaz, 2024-10-20:
Any windows devs want to spearhead this to improve UX for windows users who use remote cloud storage (i.e. one drive) ?

tbird1965, 2024-10-22:
I have noticed the backup functionality of FreeCAD saves the backup file without any issues with a junction.  If I open a file via a junction path and try to save it, I get the error, but as I work, the autosave function works just fine.  I checked my folder and the FCbak is current.  I would think the backup and save routine would use the same process.  Maybe this helps.

luzpaz, 2024-10-22:
> I have noticed the backup functionality of FreeCAD saves the backup file without any issues with a junction.

Great discovery. Any other windows users can confirm. We may have a simple diagnosis here.

luzpaz, 2024-10-22:
@tbird1965 can you share your full [About](https://wiki.freecad.org/About) info? Are you using a qt6 build ?

tbird1965, 2024-10-22:
I am using the Windows installer from the website: https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-WIN-x64-installer-1.exe.  Previously, I was using version 0.21.1 from the website.  Both act the same.

From the about, it appears to be a QT build:

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * sheetmetal 0.4.0
[/code]

luzpaz, 2024-10-23:
@tbird1965 are you planning to migrate to 1.0 ? We have an RC2 (2nd Release Client) we're recommending users to use. Can you reproduce the autosave phenomena on RC2 ?

https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

tbird1965, 2024-10-23:
Problem appears to be solved in 1.0 RC2.  I did not receive any errors saving using a junction on my Win10.  Thank you.

luzpaz, 2024-10-23:
@tbird1965 can you please share your full [About](https://wiki.freecad.org/About) info of the RC2 you tested on? TIA

tbird1965, 2024-10-23:
About still shows Qt 5:
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * sheetmetal 0.4.0
```

luzpaz, 2024-10-23:
> The only difference on my system would be

Actually there are a host of differences:  

0.21.1: `Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3`
RC2: `Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2`

So it looks like Junction is at least supported by Qt 5.15.13

This is good news. 

cc @adrianinsaval

luzpaz, 2024-10-23:
Closing as an upstream issue solved in Qt 5.15.13

closed 2024-10-23 by luzpaz