FreeCAD/FreeCAD #10849: [Problem] UI isn't uniformly the same which can cause issues and make it hard to find bugs.

MisterMakerNL, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As noted on the forum that Linux has different styles and seems to that Freecad uses those styles to style some parts of Freecad.
The issue with this is that next to having to get the same UI bug free across Linux Windows and Mac, we seem to have[ 5 different styles with Linux. ](https://doc.qt.io/qt-5/qtquickcontrols2-styles.html)
And I don't see it feasible to develop an consistent UI across all those styles.
And there fore I would suggest to set the style to a certain version.
Fusion seems to be noted the most, as in [Obelisk forum ](
https://forum.freecad.org/viewtopic.php?t=81122 )post as in [other forum posts](https://stackoverflow.com/questions/29372170/qmenu-change-background-color).
Only line of code that should be added is as far as I see it this:
`app = QApplication(sys.argv)
app.setStyle(QStyleFactory.create('fusion'))
more info here:
https://stackoverflow.com/questions/39201227/how-to-change-style-in-pyqt5

### Full version info

```shell
Not in latest dev. 0.22
Linux only.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-09-28:
Does this mean that FreeCAD won't necessarily use the same styling as the other Apps on my linux box?  
I'd rather have consistency across my environment than consistency between Win and linux.

MisterMakerNL, 2023-09-28:
We maybe only want to set this when a stylesheet is applied. Since it seems to change some tiny things that the stylesheets is missing or isn't able to change. Really subtle stuff.
So set too default when no stylesheet is applied and to something like fusion when a stylesheet is applied.
This would hopefully guarantee that all stylesheets look the same always and that no stylesheet is system default.

luzpaz, 2023-09-28:
Can we have logic that checks to see what Linux Qt theme is enabled and then let the user know that currently FreeCAD UI is optimized toward Qt Fusion theme or something ?

yorikvanhavre, 2023-09-29:
I think we can of course change the default qt style to be the one we think gives users the best experience. But we should never prevent users to apply the qt style they want. There are, apart from those 5, many additional theme engines that can be used, and that's the foundation of desktop theming. Preventing that would attract the fury of a lot of KDE users ;)

luzpaz, 2023-09-29:
@yorikvanhavre agreed. I wasn't referring to prevention but more a head-up (spitballing here)

*Hey we're seeing your opting to use the builtin Qt blah theme. JFYI, FreeCAD is optimized for Fusion Qt theme (atm). Any UI/UX issues you encounter maybe partially to blame because of this issue. Please be mindful of that if you open a ticket in this regard.*

yorikvanhavre, 2023-10-02:
Yes that might be a good way forward.. In any case I would rethink this when we actually have a consistent "preferred" Qt theme to direct users to

MisterMakerNL, 2023-10-02:
It's also good for stylesheet makers, if they develop it in mac it will probably have issues in windows. Setting style would prevent this.
One bug is the icon space in pulldown menu it appears in fusion style but not in windows style.
If you apply Fusion in Windows it also appears.

MisterMakerNL, 2023-10-03:
Ok did some digging here found a piece of code to show which styles are available:
Added it too Obelisk macro:
```
class QtNormalizer:
    from PySide import QtWidgets
    QtWidgets.QApplication.setStyle('windows')
    print(QtWidgets.QStyleFactory.keys())
    print("FreeCAD Dark Success!")
```
And @chennes did it on Mac
['macintosh', 'Windows', 'Fusion']
I did it on windows:
['windowsvista', 'Windows', 'Fusion']
And on Linux Mint:
 ['Windows', 'Fusion']
So basically we have 2 options: Fusion and Windows.
Now I compared the 2 on Linux Mint and put it in a gif to see the difference:
[image]
and on my Windows 10 PC:
[image]

And now I am a bit baffled, like the biggest difference is the pulldown menu's. But also a lot of tiny padding differences.
What to do now?
Add a style switcher in the menu?
Try to make the stylesheets work for the both?
Just force one of the 2 with stylesheets?

luzpaz, 2023-10-03:
Thanks for this really thorough research, @MisterMakerNL !

MisterMakerNL, 2023-10-05:
> Probably we can "iron out" these differences in a stylesheet... That could be good

Yeah after thinking about it this is probably the best solution.

obelisk79, 2023-10-06:
I disagree. FreeCAD has a big problem in its ability to present a uniform appearance across platforms. Qt Widget interfaces are quite poor at displaying the interface uniformly across the draw styles. The existing stylesheets are fulls of OS specific bits that are poorly documented and are intended to try and normalize appearance regardless of Qt's default widget style. This becomes difficult to manage.

FreeCAD in my opinion should define a specific draw style internally and then allow for adjustment via a property in the property editor for diehards who are into extreme customization. This would make the visual appearance of FreeCAD more uniform across all platforms AND make management of such appearance (the stylesheet) much simpler to maintain, update and modify since the project would only officially need to support one style that presents uniformly regardless of OS or DE.

Trying to make a stylesheet look the same on all platforms is the complex approach  akin to hammering a square peg into a round hole. With enough effort you can succeed but it is not the most practical approach.

I propose 'Fusion' as it is available on *all* platforms and unlike the 'windows' style it doesn't ring of negative connotations that many people associate with Microsoft.

I believe that @MisterMakerNL has a concern about the QComboBox style, which admittedly looks a little better in Windows style. However much of the rest does not. In the case of that, it would be possible to custom class QComboBox with an override for Windows style, but I doubt doing such a thing would be practical.

obelisk79, 2023-10-06:
To be clear, what could be defined in 5 lines of code, would take several hundreds of lines of qss to reproduce.

MisterMakerNL, 2023-10-06:
I don't want to make the fusion combobox work the same. I see there some bugs in fusion that aren't in windows. One bug is the empty icon soace holder.

obelisk79, 2023-10-06:
Fair enough, can you share screenshots to show the difference because I'm not familiar with the bug you're referring to.

MisterMakerNL, 2023-10-06:
First one done:
https://github.com/FreeCAD/FreeCAD/pull/10954
Seems like the biggest fish too!

MisterMakerNL, 2023-10-07:
I cleaned up the PR but for that made a new one, I also I might overreacted a bit here on the issue. I think fixing Qcombobox is probably the only issue here, the rest is just padding and for size. Stuff normal users won't even notice and doesn't impact usability. So I am closing it. 
New PR https://github.com/FreeCAD/FreeCAD/pull/10968

closed 2023-10-07 by MisterMakerNL