FreeCAD/FreeCAD #10851: [Problem] Freecad tabs to a different file whenever I start a sketch related action

AlphaDragon601, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm not certain how long his has been happening, maybe I just didn't notice it after updating, but when I'm working with more than one file open I've found that doing almost anything related to a sketch (opening it to modify it, padding it, pocketing it,etc) the task menu will open with what I selected but for some reason my active window tabs to the other file I have open. I then have to tab back over to actually finish what I was doing. Nothing grander has happened but its rather annoying when I want to keep another model open for reference and I keep getting derailed from what I'm doing. 

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.56
  * Marz 0.0.27-alpha
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/50785295/4793fca2-e2b9-49d3-98dd-0fb8cdbf1e41

https://github.com/FreeCAD/FreeCAD/assets/50785295/8de38c7c-6205-44a4-929b-c219e97fb3fb

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-28:
Is this related to #7745 ?

AlphaDragon601, 2023-09-28:
> Is this related to #7745 ?

Sounds like the same issue, though I'm not on Mac.

xtemp09, 2023-09-28:
I cannot reproduce on 

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34310 (Git)
Build type: Release
Branch: master
Hash: 3e789dfc85c84e5e6295df908795db5830d28d43
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

and 

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34438 (Git)
Build type: Debug
Branch: master
Hash: 247df9939108ebe6cf398868e59f49d965c3e381
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

though, I don't know the exact steps to reproduce the problem.

AlphaDragon601, 2023-09-29:
Can't reproduce the problem after a reboot, toggled the auto switch setting as well. Either the problem was related to that specific session and the reboot fixed it or theres some missing detail in my repro efforts. The setup I had upon the issue creation was 2 files and there spreadsheets open in tandum. I set this up again with the same models and made sure all bodies were shown but nothing.

closed 2023-09-29 by AlphaDragon601