FreeCAD/FreeCAD #10861: [Problem] Convert STP file to gltf, it works, but miss color information

gt2ming, 2023-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello!

When I convert the [.stp] file to [.gltf], it miss color information.

I referred to this issue, and search the forum, but I can't find a solution.

https://github.com/FreeCAD/FreeCAD/pull/8649#issuecomment-1728684497

this is my code

```python
import os
import sys
sys.path.append('D:\\Program Files\\FreeCAD 0.21\\bin')
import FreeCAD
import Import
import Mesh
import Part
import MeshPart

if __name__ == '__main__':
  stpFilePath = 'D:\\pcb1.stp'
  gltfFilePath = 'D:\\pcb1.gltf'
  objectList = []
  result = Import.open(stpFilePath)
  objectList.append(FreeCAD.getDocument("Unnamed").getObject("Model"))
  objlist = [i[0] for i in result]
  for obj in objlist:
    data = obj.Shape.tessellate(0.5)
  Import.export(objectList,gltfFilePath)
```

Display the original model in freecad, Or export the [.gltf] in FreeCAD gui, it has color.

[image]

The result [.gltf] when I export by the python script, and view in [Three.js], it hasn't color

[image]

Thank you very much!

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-29:
Are you certain this isn't an issue with threejs ?

Ping @DanielOrchard @morganrallen (other users of gltf) do you mind confirming this ?

morganrallen, 2023-09-29:
It's been a while since I tried to export gltf from python but as I recall this is the same result I got. I'm away from home at the moment but will confirm tomorrow. Off the top of my head this is because all color information is stored in FreeCADGui and therefore unavailable to the headless script.

luzpaz, 2023-09-29:
> all color information is stored in FreeCADGui and therefore unavailable to the headless script

Maybe we should change the ticket name in that case ?

wwmayer, 2023-09-29:
This is a regression.

In the past it has worked to export geometry and color in headless mode but now it fails.
This is the related forum thread: https://forum.freecad.org/viewtopic.php?p=33331#p33331
And this is the change that supported it: 64890a45525d363

Another forum thread is:
https://forum.freecad.org/viewtopic.php?f=22&t=23494&p=187294#p187294
and an improvement was done with 02ab1c9424944b

The regression happened with: 32ad54c9eff3b47

wwmayer, 2023-10-02:
With the pushed branch export of glTF files now support colors. All what you have to do is to replace the last line with
`Import.export(result,gltfFilePath,legacy=True)`

This means that currently only the legacy export function supports colors again. The next step is to make this available for the new export function.

wwmayer, 2023-10-02:
The second part of the PR makes this function available for the new export function in headless mode.

closed 2023-10-02 by wwmayer

morganrallen, 2023-10-02:
Thanks @wwmayer

commit 74e658c7 (2023-10-02): Import: improve new STEP/IGES/glTF export function to support colors per face in headless mode

Fixes #10861