FreeCAD/FreeCAD #10865: [Problem] CalculiX binary provided with Windows weekly builds causes an issue, original binary doesn't

FEA-eng, 2023-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The attached input file (change the extension to .inp) was generated in the FreeCAD FEM workbench and modified using a built-in editor. It doesn't work when submitted using the CalculiX binary attached to weekly builds. Basically, this example doesn't work in CalculiX 2.20 but works in 2.21. The ccx binary provided with the weekly builds on Windows seems to be 2.21 but this example still doesn't work with it. However, it works when submitted using the original binary from http://www.dhondt.de/calculix_2.21_4win.zip

File: [FEMMeshGmsh.txt](https://github.com/FreeCAD/FreeCAD/files/12764521/FEMMeshGmsh.txt)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git)
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-29:
libpack or conda generated ?

FEA-eng, 2023-09-29:
> libpack or conda generated ?

It's the Conda one for Windows.

luzpaz, 2023-11-08:
@adrianinsaval is this on your radar ?

adrianinsaval, 2023-11-09:
what do you mean when you say "doesn't work" I'm not familiar with how calculix is used so I need you to be more specific. What steps are you folowing, what errors are you encoutering?

FEA-eng, 2023-11-09:
> what do you mean when you say "doesn't work" I'm not familiar with how calculix is used so I need you to be more specific. What steps are you folowing, what errors are you encoutering?

Unfortunately, there's no error message in this case. CalculiX sometimes just fails to complete the analysis without generating any error messages.

You can put this input file (with extension changed from .txt to .inp) in the bin directory of FreeCAD and run it from the command line by typing `ccx FEMMeshGmsh`

It will terminate prematurely like this:

```
...
   orientations:            0
   amplitudes:            2
   data points in all amplitudes:            2
   print requests:            1
   transformations:            0
   property cards:            0

 STEP            1
```

When submitted using the ccx_static.exe binary from the website linked above, it runs properly and ends with a message like this:

```
Total PaStiX Time: 0.356278
CCX without PaStiX Time: 0.366578
Share of PaStiX Time: 0.492875
Total Time: 0.722856
Reusability: 0 : 1
________________________________________

 Using up to 8 cpu(s) for the stress calculation.

 Job finished

________________________________________

Total CalculiX Time: 0.799191
________________________________________
```

FEA-eng, 2025-01-26:
Still relevant

luzpaz, 2025-01-26:
What's the fix here ?

FEA-eng, 2025-01-26:
> What's the fix here ?

We would need a discussion about the ccx binary provided with FreeCAD - if it can't be just the official binary from the website, what DLLs to include (also because of this: https://forum.freecad.org/viewtopic.php?t=94168) and so on.

FEA-eng, 2025-09-17:
Even if not for this particular analysis, the topic is still very relevant in general. CalculiX binaries provided with FreeCAD cause several issues while the ones downloaded from the official website work fine.

adrianinsaval, 2026-04-17:
I submitted the PR on conda-forge to get #18368 fixed: https://github.com/conda-forge/calculix-feedstock/pull/34 once that lands, is there any other problem with the conda-forge binary? I would prefer to see it fixed

oursland, 2026-09-22:
This should be resolved with https://github.com/FreeCAD/FreeCAD/pull/32866

FEA-eng, 2026-09-23:
It still terminates prematurely for me when I run the attached input file with the ccx binary attached to the latest weekly (FreeCAD_weekly-2026.09.23-Windows-x86_64\bin\ccx.exe)

oursland, 2026-09-23:
@FEA-eng What's your locale and language?  There's been several changes both upstream in my patch submissions to upstream to address problems with Windows and non-en_US locales.

FEA-eng, 2026-09-24:
> [@FEA-eng](https://github.com/FEA-eng) What's your locale and language? There's been several changes both upstream in my patch submissions to upstream to address problems with Windows and non-en_US locales.

My locale / Windows language is Polish, but I always use FreeCAD in English. The issue with the decimal separator (#18368) was locale-dependent (also happened for German users). But this one seems different.

oursland, 2026-09-27:
An analysis shows that this is due to a 2D axisymmetric model that fixes DoFs 1-6, however DoFs 4-6 are rotations that are not present.  CCX is internally turning this 2D mesh into 3D wedge.  In processing this it doesn't fill in all of the necessary loop variables, resulting in reading memory that has not been initialized.

What apparently is happening is that in some builds, these leftover random memory doesn't crash the system, but in the builds on conda-forge they do.  In all instances, there should not be reading these uninitialized values.

I have a PR prepared for upstream on this issue, but I'm now performing sweeps with AddressSanitizer and valgrind to see if we can identify yet more areas in which there are genuine defects in the code (mostly the Fortran code) that need to be resolved.

oursland, 2026-10-05:
@FEA-eng Have you had an opportunity to test with a recent weekly build?

FEA-eng, 2026-10-05:
> [@FEA-eng](https://github.com/FEA-eng) Have you had an opportunity to test with a recent weekly build?

It seems to be working now. But doesn't it require fixing the erroneous handling of redundant DOFs on the CalculiX side first?

oursland, 2026-10-05:
@FEA-eng The fixes are being upstreamed and will be present in the next release of CalculiX.

* https://github.com/Dhondtguido/CalculiX/issues/187

Their process is odd and not at all advised, but instead of merging PRs, they copy the code and close the PR.  So the closed PRs in that list are actually accepted, but not properly attributed.

FEA-eng, 2026-10-06:
> [@FEA-eng](https://github.com/FEA-eng) The fixes are being upstreamed and will be present in the next release of CalculiX.

Yeah, but is the fix already included in the CalculiX binary provided with the new weeklies (since it seems to be working somehow)?

oursland, 2026-10-06:
Yes, the fixes are already included.