FreeCAD/FreeCAD #10868: [Problem] crash on access of null Part.Edge Parameters

alexneufeld, 2023-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

paste the following to the python console:
``` python
import Part
a = Part.Edge()
print(a.FirstParameter)

```
this crashes FreeCAD:
``` bash
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3e710) [0x7f94ab63e710]
#1  0x7f9477d775da in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /usr/lib/libTKBRep.so.7+0x2a
#2  0x7f9477d871a6 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /usr/lib/libTKBRep.so.7+0x86
#3  0x7f9477d89699 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /usr/lib/libTKBRep.so.7+0xd9
#4  0x7f947931ea5b in Part::TopoShapeEdgePy::getFirstParameter() const from /usr/lib/freecad/lib/Part.so+0x5b
#5  0x7f94793191e6 in Part::TopoShapeEdgePy::staticCallback_getFirstParameter(_object*, void*) from /usr/lib/freecad/lib/Part.so+0x36
#6  /usr/lib/libpython3.11.so.1.0(_PyObject_GenericGetAttrWithDict+0x25d) [0x7f94ad5f82bd]
#7  0x7f94adb52f72 in Base::PyObjectBase::_getattr(char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0xa2
#8  0x7f94adb54b00 in Base::PyObjectBase::__getattro(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADBase.so+0xa0
#9  /usr/lib/libpython3.11.so.1.0(PyObject_GetAttr+0x52) [0x7f94ad5dd252]
#10  /usr/lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0xf4a) [0x7f94ad5e4cba]
#11  /usr/lib/libpython3.11.so.1.0(+0x29e49a) [0x7f94ad69e49a]
#12  /usr/lib/libpython3.11.so.1.0(PyEval_EvalCode+0xac) [0x7f94ad69deac]
#13  0x7f94aeaab62a in Gui::InteractiveInterpreter::runCode(PyCodeObject*) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x6a
#14  0x7f94aeaaba84 in Gui::InteractiveInterpreter::runSource(char const*) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x64
#15  0x7f94aeaad68c in Gui::InteractiveInterpreter::push(char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xbc
#16  0x7f94aeab67e4 in Gui::PythonConsole::runSource(QString const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x104
#17  0x7f94aeab6d1a in Gui::PythonConsole::keyPressEvent(QKeyEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x36a
#18  0x7f94acdaee6d in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x55d
#19  0x7f94ace5d703 in QFrame::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x23
#20  0x7f94acd7893f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8f
#21  0x7f94acd7d958 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0xc68
#22  0x7f94ae8b5b09 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#23  0x7f94ac09c6f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#24  /usr/lib/libQt5Widgets.so.5(+0x1cec5b) [0x7f94acdcec5b]
#25  0x7f94acd7893f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8f
#26  0x7f94ae8b5b09 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#27  0x7f94ac09c6f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#28  0x7f94ac545570 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /usr/lib/libQt5Gui.so.5+0xf0
#29  0x7f94ac52a6c5 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Gui.so.5+0xb5
#30  /usr/lib/libQt5XcbQpa.so.5(+0x653e0) [0x7f94a60fc3e0]
#31  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x141) [0x7f94aa310a61]
#32  /usr/lib/libglib-2.0.so.0(+0xb7c29) [0x7f94aa36dc29]
#33  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x32) [0x7f94aa30e112]
#34  0x7f94ac0eb51c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x6c
#35  0x7f94ac09b404 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x134
#36  0x7f94ac09c8a3 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x93
#37  0x7f94ae82715b in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x1bab
#38  freecad(+0x4e13) [0x56402bac8e13]
#39  /usr/lib/libc.so.6(+0x27cd0) [0x7f94ab627cd0]
#40  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7f94ab627d8a]
#41  freecad(+0x5fd5) [0x56402bac9fd5] 
```

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Curves
  * python_site_packages
  * Assembly3 0.12.0
  * robotpy
  * brlcad_wb
  * fasteners 0.4.62
  * FreeCAD-PreferencePacks.backup1669700355.378573
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.0.0
  * Assembly4
  * FreeCAD_SheetMetal 0.2.56
  * Silk
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-30:
Can reproduce on 

```bash
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git) AppImage
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

```shell
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3e710) [0x7f562260e710]
#1  0x7f5566d57ea6 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCANuUdH2/usr/lib/./libTKBRep.so.7+0x26
#2  0x7f5566d69a01 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /tmp/.mount_FreeCANuUdH2/usr/lib/./libTKBRep.so.7+0xd1
#3  0x7f5566d6ac15 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /tmp/.mount_FreeCANuUdH2/usr/lib/./libTKBRep.so.7+0xd5
#4  0x7f556834efa7 in Part::TopoShapeEdgePy::getFirstParameter() const from /tmp/.mount_FreeCANuUdH2/usr/lib/./Part.so+0x57
#5  0x7f556834f062 in Part::TopoShapeEdgePy::staticCallback_getFirstParameter(_object*, void*) from /tmp/.mount_FreeCANuUdH2/usr/lib/./Part.so+0x32
#6  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_GenericGetAttrWithDict+0x232) [0x7f5624747172]
#7  0x7f5623d81284 in Base::PyObjectBase::_getattr(char const*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADBase.so+0x84
#8  0x7f5623d8173c in Base::PyObjectBase::__getattro(_object*, _object*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADBase.so+0x9c
#9  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libpython3.10.so.1.0(PyObject_GetAttr+0x52) [0x7f56247460b2]
#10  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4952) [0x7f562473d2c2]
#11  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libpython3.10.so.1.0(+0x1de9b0) [0x7f56247de9b0]
#12  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libpython3.10.so.1.0(PyEval_EvalCode+0x8c) [0x7f56247de8fc]
#13  0x7f5625157e01 in Gui::InteractiveInterpreter::runCode(PyCodeObject*) const from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0xc1
#14  0x7f5625158480 in Gui::InteractiveInterpreter::runSource(char const*) const from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0x60
#15  0x7f56251585a8 in Gui::InteractiveInterpreter::push(char const*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0xb8
#16  0x7f56251587d6 in Gui::PythonConsole::runSource(QString const&) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0x116
#17  0x7f5625159952 in Gui::PythonConsole::keyPressEvent(QKeyEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0x492
#18  0x7f5623794282 in QWidget::event(QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Widgets.so.5+0x1b4
#19  0x7f56237fe7d7 in QFrame::event(QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Widgets.so.5+0x27
#20  0x7f562376a087 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#21  0x7f562376e0cc in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Widgets.so.5+0x3c2
#22  0x7f5624ef6fd9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0x99
#23  0x7f5622bc2a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Core.so.5+0x8e
#24  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Widgets.so.5(+0x1a44f5) [0x7f56237a44f5]
#25  0x7f562376a087 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#26  0x7f5624ef6fd9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0x99
#27  0x7f5622bc2a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Core.so.5+0x8e
#28  0x7f56231073e3 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Gui.so.5+0xcd
#29  0x7f56230f6091 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Gui.so.5+0x57
#30  /tmp/.mount_FreeCANuUdH2/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5672c) [0x7f561dbce72c]
#31  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/./libglib-2.0.so.0(+0x5a3e7) [0x7f56211113e7]
#32  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/./libglib-2.0.so.0(+0x5d397) [0x7f5621114397]
#33  /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f562111498e]
#34  0x7f5622bf3d50 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Core.so.5+0x5c
#35  0x7f5622bc05d8 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Core.so.5+0x180
#36  0x7f5622bc47af in QCoreApplication::exec() from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libQt5Core.so.5+0x107
#37  0x7f5624e5d7f4 in Gui::Application::runApplication() from /tmp/.mount_FreeCANuUdH2/usr/bin/../lib/libFreeCADGui.so+0x1f14
#38  /tmp/.mount_FreeCANuUdH2/usr/bin/freecad(+0x8620) [0x55574a0f0620]
#39  /usr/lib/libc.so.6(+0x27cd0) [0x7f56225f7cd0]
#40  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7f56225f7d8a]
#41  /tmp/.mount_FreeCANuUdH2/usr/bin/freecad(+0x8981) [0x55574a0f0981]
```

xtemp09, 2023-09-30:
No crash on Windows:

```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
Base.FreeCADError: Unknown exception while reading attribute 'FirstParameter' of object 'TopoShape'
```

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git)
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

luzpaz, 2023-10-01:
Anyone volunteer to fix this ?

tomate44, 2023-10-01:
Confirmed

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34473 (Git)
Build type: Release
Branch: master
Hash: 5dfd4d519068199f92c6bfd0431c0b99872ede0b
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Trails 2022.1.0
  * estimateWB 0.1.0
  * CurvesWB 0.6.13
  * Help 1.0.3
  * Render 2023.8.18
  * lattice2 1.0.0
  * sheetmetal 0.3.0
  * workfeature
  * ProDarkThemePreferencePack 1.0.0
  * pivy_trackers
  * PT 0.5.11
```

The crash happens on all the calls to BRepAdaptor_Curve, even inside a try block.
So, many methods are affected, not only getFirstParameter().

I don't know what is the most appropriate way to fix this, though.

tomate44, 2023-10-01:
Same bug on Part.Face : 
 ```
import Part
f = Part.Face()
f.valueAt(0, 0)
 ```
crashes on calling BRepAdaptor_Surface

luzpaz, 2023-10-03:
@wwmayer looks like a systemic issue. Care to weigh-in at your convenience ?

FlachyJoe, 2023-10-03:
Shape should be tested before accessing it's members

```cpp
Py::Float TopoShapeEdgePy::getFirstParameter() const
{
    const TopoDS_Edge& e = TopoDS::Edge(getTopoShapePtr()->getShape());
    if (e.IsNull()) {                                         // Add a validity check
        PyErr_SetString(PyExc_ValueError, "Edge is null");    //
        //...
    }
    BRepAdaptor_Curve adapt(e);
    double t = adapt.FirstParameter();
    return Py::Float(t);
}
```

or with a `try...catch` as in TopoShapeEdgePy::curveOnSurface

commit 74a31d14 (2023-10-05): Part: fix #10868 crash with null edge
closed 2023-10-05 by wwmayer