FreeCAD/FreeCAD #10869: [Problem] Part Measure toolbar could be less cluttered.

PaddleStroke, 2023-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently there are 7 icons in the Part Measure toolbar. I think it can be improved by making a group command with a widget like the sketcher grid tool for example.
[image]

I think the toolbar could be shrinked to a single tool with a dropdown to show a widget with reset/refresh buttons, and toggle checkboxes.

Having said that I don't know if it's worth the effort as perhaps time would be better invested into unifying measure tools into a single tool in the view toolbar. And someone is already working on that I think.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-09-29:
Keep in mind the GSOC project "Unified measurement tool". Ideally the new measure tool will only need one icon.

luzpaz, 2023-09-30:
Can we close this in favor of #8561

PaddleStroke, 2023-09-30:
Yes sure

closed 2023-09-30 by PaddleStroke