FreeCAD/FreeCAD #10870: Custom installation path parameter doesn't work during install

CameronGo, 2023-09-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When installing the application as a user, using the install parameter to install to a custom installation path does not appear to work. For example, when I run this command: `FreeCAD-0.21.1-WIN-x64-installer-1.exe /S /D=%LOCALAPPDATA%\Programs\FreeCAD\`

The application continues to install to the default installation folder: `%LOCALAPPDATA%\Programs\FreeCAD 0.21`

I want to control the installation path so that I can remove the version number from the folder structure and allow our endpoint management software to effectively detect the presence of an older version and replace it with a newer version.

### Full version info

```shell
0.21.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-18:
@chennes :point_up:

metaad, 2024-02-14:
is there any update on the issue.

maxwxyz, 2024-06-13:
@chennes is this still relevant?

CameronGo, 2025-02-04:
I've just run the same install command referenced in my original post and get the same result with no change, so yes to me it still seems relevant.

CameronGo, 2025-09-30:
Still relevant