FreeCAD/FreeCAD #10874: [Problem] File extension .FCStd is not standard, proposing a change to .fcstd as backup then .fcad as new standard

zeeb1337, 2023-09-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The file extension .FCStd should be changed to .fcstd to become standardized like every other file extension that is used internationally.

Another proposal from the Development section in the Discord channel of FreeCAD is to make .fcstd a backup extension then add a completely new project file extension as standard such as .fcad.

This looks more professional and clean, easier to remember and poses less potential problems with for example advanced searching where capital letters play a role, while also being backwards compatible with already created projects that is using the .FCStd or .fcstd extension.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Swedish/Sweden (sv_SE)
[/code]
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-09-30:
FWIW, Microsoft specifically says that an extension is "three or four letter": https://support.microsoft.com/en-us/windows/common-file-name-extensions-in-windows-da4a4430-8e76-89c5-59f7-1cdbbc75cb01

Of course, in that same document they then proceed to list out "common extensions" which includes several longer ones!

I was unable to find any actual standard, everything seems to be just de facto: does XDG have anything to say on the matter?

WandererFan, 2023-09-30:
I like it.  Read *.fcstd and *.fcad.  Write *.fcad (unless overridden?). 

When you say "backup extension",  I understand you to mean "an additional acceptable extension", and not "a replacement for fcstd1 or fcbak autosave files".  Correct?

zeeb1337, 2023-09-30:
That is correct, the project should then by default save project files onward as .fcad files instead of .fcstd files, but will also accept .fcstd files.

macdroid53, 2023-09-30:
> FWIW, Microsoft specifically says that an extension is "three or four characters": https://support.microsoft.com/en-us/windows/common-file-name-extensions-in-windows-da4a4430-8e76-89c5-59f7-1cdbbc75cb01
> 
> Of course, in that same document they then proceed to list out "common extensions" which includes several longer ones!
> 
> I was unable to find any actual standard, everything seems to be just de facto: does XDG have anything to say on the matter?

Isn't this a left over from the days when on microsoft os's you couldn't have more that 3 chars in an extension?

zeeb1337, 2023-09-30:
https://superuser.com/questions/315393/there-is-a-length-limit-to-windows-file-extensions

> While you are free to use any length of extension you wish, I would not recommend using a very lengthy one for one reason: _convention_. Most file extensions are three to four alphanumeric characters. Anything longer, or with funny characters, is going to 'stand out' quite a bit more, be harder to memorize, and may be less likely to be adopted (again, it all depends). For a list of common file extensions.

macdroid53, 2023-10-01:
Is it not the default in Windows (and Mac?) to hide the extension in the associated file browser?

Can you provide some documentation that defines the conventions you think are used?

Can you provide discussion outside the FreeCAD sphere that  define the convention?

Was there not a change to the FreeCAD file extension sometime during the 0.18 (or maybe 0.19??) dev cycle?

What was the motivation behind that change?

xtemp09, 2023-10-01:
If yorikvanhavre changes the name of the master branch, I propose the following extension to show that FreeCAD supports wokeness: 🛠

Just imagine: my_project.🛠

Or other emojis, like: first_attic.👩‍💻 to show that women too can use computers.

Looks good, right? Backups could be: case_N28.🛅 or case_N28.💼

zeeb1337, 2023-10-01:
@macdroid53 https://developers.google.com/style/filenames then you can see common conventions at the very bottom of the page, but it's all a good read.

macdroid53, 2023-10-01:
The Wikipedia entry notes:
"Eventually, Windows 95 introduced support for long file names, and removed the 8.3 name/extension split in file names from non-NT Windows,...but it removed the important length restriction and allowed files to have a mix of upper case and lower case letters, on machines that would not run Windows NT well."

The google link is the Google Developers style guide. The list you refer to is prefaced with "...table lists some examples of filename extensions...". 

There are 3 and 4 letter extensions shown.

Blender writes a .blend, Solidworks .sldprt, .sldasm, .slddrw, Altium has .BomDoc, CmpSync, etc.

I see no compelling argument.

Ultimately, I have no authority over this or any other feature of FreeCAD so, I'll bow out of this discussion now.

zeeb1337, 2023-10-01:
Do share why you would like to keep .FCStd instead of switching to .fcstd and .fcad.

macdroid53, 2023-10-01:
> @macdroid53 Do share why you would like to keep .FCStd instead of switching to .fcstd and .fcad and why it would be better to keep it as is. Also why it's important to keep the capital letters in the extension since that is what you're against removing even when you've been informed about the potential issues that comes with it.
> 
> It would be interesting to see the argument behind the importance in keeping it, especially when it will stay backwards compatible to ensure existing projects don't suffer from the change. Meaning your and everyone else's project files will work as usual if this change were to be implemented.

I have yet to be informed of "the potential issues".

You write in the issue statement: "...Capital letters in file extensions is shunned by a great many due to various issues that can arise from it..."

What are the "various issues" with capital letters in file extensions? Altium Designer is used by many well known industry names (BAE systems, Microsoft, IRobot, Texas Instruments, etc.). Altium uses upper/lower case file extensions with no repercussions (loss of revenue, etc.)

Please provide specific examples of these potential issues.

kadet1090, 2023-10-01:
Altium Designer is Windows Only, Windows uses NTFS file system which is case insensitive and does not really care. In POSIX filesystems that are often case sensitive it is often painful to deal and debug issues when files have different letter casing. There were many instances where I had to debug stuff which worked on developers computer because he uses windows and named file with `.PNG` instead of `.png` and for NTFS it worked and on EXT4 it did not - it was "fun" debug time. 

There are at least two documents linked here that are against using capital letters in filenames - so if there is no good reason to use it I think that we should change it.

zeeb1337, 2023-10-01:
> I have yet to be informed of "the potential issues".

@macdroid53 Advanced searching where capital letters play a role. As I wrote in my original post and the post on the Discord channel which you have read as well.

So there we have an example of a potential issue with the convention. Now it would be good if you provide your reasoning for not wanting the capital letters removed. Aside from personal aesthetic preferences, I trust you have a technical reason as to why it should stay as is?

macdroid53, 2023-10-01:
No, I have never argued there was a technical reason to keep it.

I argue that is is "make work" with a low benefit. I said as much in the Discord thread.

Also, IMO, "Advanced searching where capital letters play a role." is nebulous. What is an example where this would be an issue?  If you enter any of these strings FCSTD, .FCStd, .fcstd in a search engine the same result are presented. And those results are that it is a FreeCAD file.

chennes, 2023-10-01:
It may be of minimal benefit, but then again, if it's something that interests @zeeb1337, I don't want to discourage a new contributor from joining the ranks (and possibly it might be a gateway to further contributions 😁 ). I see no downsides to migrating to an all-lowercase version of the fcstd extension. Switching extensions entirely is likely more controversial, but personally I wouldn't mind one that was more "pronounceable". It doesn't look like anyone is using "*.fcad", so I'm game for adding it as one of our allowed extensions, and possibly even making it the "preferred" extension. @FreeCAD/maintainers any other opinions?

zeeb1337, 2023-10-01:
Whoever makes this contrib, I'm confident that it is in the best interest of the FreeCAD community and the project as a whole. I see only benefits with this change.

zeeb1337, 2023-10-01:
@macdroid53 > I argue that is is "make work" with a low benefit.
That should not be a concern, contributors are free to decide what issues they want to work on and how they prioritize it. One day they might only have for example 1 hour they can spend on open-source issues so they might decide to fix a small issue. Another day they might have 6 hours or more and decide to tackle something bigger.

There is nothing forcing a contributor to tackle a certain problem, every contributor is free to choose as they please. Contributors who share your point of view simply won't tackle an issue such as this and thus will never "waste" time on it but instead focus their attention elsewhere. So there is no cause for concern.

adrianinsaval, 2023-10-02:
I don't agree with changing the extension itself, since it gives no real benefit. But I do agree that we should be using lowercase extension instead. Having extensions with mixed case has been problematic with links for some people. I can't find a github issue right now but this was brought up by zolko before.

Roy-043, 2023-10-02:
I do not have an opinion about the actual file extension, but would like to point out that any change will have an impact on the Wiki documentation.

wwmayer, 2023-10-02:
At the beginning FreeCAD was inspired by CATIA and because it uses the extensions .CATPart, .CATProduct, ... I think this was also the reason why FreeCAD used .FCStd instead of .fcstd.

Moving to lowercase letters will cause some trouble on Linux with existing files because the native file dialogs usually are case-sensitive and when switching to .fcstd it won't display files with the .FCStd extension any more.

Moving to a different extension like .fcad doesn't make any sense to me. IMO, it doesn't have any benefits but will cause a lot of confusion instead.

> FWIW, Microsoft specifically says that an extension is "three or four characters":

MS Visual Studio: vcproj, vcxproj, ...

yorikvanhavre, 2023-10-02:
It's funny because at some point in time the FreeCAD linux version DID write .fcstd (lower-case). Then it was fixed to match the behaviour of other platforms.

I would think extensions should always be case-insensitive. This is very common with DWG files, there are still an awful lot of apps out there that writes xxx.DWG files. So writing .FCStd is just a "cosmetic" way to represent FreeCAD files. FreeCAD at least (but currently the only app to open FreeCAD files I think :cry: ), will happily open any .FCStd, .fcstd or .fcStD files, no matter the case.

Apart from adopting a more common style, I see little benefit from changing that now, unless of course there is a concrete problem with it (@Zolko-123 might know?)

Now the reflexion about .fcstd vs .fcad is a good one. I believe the original idea was to do like solidworks, have one file format for parts and another for assemblies. But will we ever do that? Will we ever have assemblies with another file extension? I have the impression we will do everything with .fcstd files forever. So indeed adopting a more generic extension

kadet1090, 2023-10-02:
>Moving to lowercase letters will cause some trouble on Linux with existing files because the native file dialogs usually are case-sensitive and when switching to .fcstd it won't display files with the .FCStd extension any more.

Is there anything stopping us from providing multiple extensions as options, QT allows that, dialogs should work. `FreeCAD File (*.fcstd, *.FCStd, *.fcad)`. For me it's simple a matter of choosing canonical extension, dialogs should support all of them regardless of choice.

xtemp09, 2023-10-02:
Does introduction of the solution for _Topological Naming Problem_ cause incompatibility between older FreeCAD versions and the new?

If it does, change of the file format along with the extension is understandable.

chennes, 2023-10-02:
> Does introduction of the solution for _Topological Naming Problem_ cause incompatibility between older FreeCAD versions and the new?

No. Older files will still be supported, they just won't have naming history in them, and older versions of FreeCAD will be able to open the new files, they will just drop the naming stuff.

yorikvanhavre, 2023-10-02:
> No. Older files will still be supported, they just won't have naming history in them, and older versions of FreeCAD will be able to open the new files, they will just drop the naming stuff.

(Off-topic but I can't resist: This is where you really see the true power of non-commercial software. NEVER would a commercial software allow that)

adrianinsaval, 2023-10-02:
> Moving to lowercase letters will cause some trouble on Linux with existing files because the native file dialogs usually are case-sensitive and when switching to .fcstd it won't display files with the .FCStd extension any more.

I don't think that is true. Currently both .FCStd and .fcstd (and even other variations like .fcStD) are shown in the dialog for me. I think most software is made to be case insensitive with the extension. Even FreeCAD mostly is, with the only exception that links that are saved pointing to `file.FCStd` fail if the linked file is now named `file.fcstd` instead.

yorikvanhavre, 2023-10-02:
So a first. immediate fix here would be to make sure all code that deals external files (App::Links and Arch References IIRC) are ignoring the case of extensions.

adrianinsaval, 2023-10-02:
that would be a good idea even if we don't switch to using lowercase by default

wwmayer, 2023-10-02:
> I don't think that is true.

This **is** true and in the past we had many complaints about that. And that's one of the reasons why I don't use the native file dialog but the Qt implementation because it ignores case-sensitivity.

I guess it depends on which desktop is used if variations of FCStd are shown or not. For KDE it may work since it's based on Qt but on my XFCE desktop the native file dialog doesn't show any variation.

zeeb1337, 2023-10-04:
Are we slapping a "verified" on this or closing it?

yorikvanhavre, 2023-10-04:
I'm reading a bit on the internet and finding out that beings case-sensitive or not is a filesystem feature. Some systems (FAT32 for ex) are case-insensitive while others are case-sensitive. So on one, SOMETHING.TXT == something.txt, on others not (which explains the "old" DWG case, it all comes from the paleolithic or so).  I suppose Qt adopts what the filesystem says...

That means FCStd == fcstd used to be true on windows, and should now on most filesystems be false. (Not sure about recent windows filesystems?)

In any case, supporting both in the above contexts seems like a good move. I'll adapt Arch already

Zolko-123, 2023-10-04:
> So a first. immediate fix here would be to make sure all code that deals external files (App::Links and Arch References IIRC) are ignoring the case of extensions.

yes, I think that that's the real problem, and not the Qt file chooser dialog. The faulty references to _FCStd_ or _fcstd_ in an _App::Link_ will cause a real error when loading a file, not a mere inconvenience when trying to open manually a file. 

This problem was seen by real users, and the problem is that the _App::Link_ stores the reference to the external file as a string to the path. Depending on the OS, the file extension can be _fcstd_ or _FCStd_, and if wrong then it throws an error. BUT: it is entirely possible that the reference is stored in lower-case _fcstd_ while the actual file has _FCStd_ extension, thus for App::Link the best behavior would be to check for the exact path, then check for _fcstd_ **AND** _FCStd_ extensions of that file. 

Also, from what I've read, the MacOS filesystem is case insensitive BUT still stores extensions as they are entered: `toto.FCStd == toto.fcstd` for the OS, but they still will be stored as different strings in the _App::Link_ property

wwmayer, 2023-10-04:
> The faulty references to FCStd or fcstd in an App::Link will cause a real error when loading a file

But at what point does it happen? Is it that the link property of the App::Link changes its content or is it that the referenced file changes its name?

> it is entirely possible that the reference is stored in lower-case fcstd while the actual file has FCStd extension

If the link property changes then this must be considered as bug. At some point the correct path of the referenced file must be determined and assigned to the link property and from that point on it mustn't change any character from lower case to upper case or vice versa.

The more complicated problem is if the name of the referenced file changes because this isn't something FreeCAD can control in any way. But I don't think that this happens automatically when copying files from one file system to another (e.g. from ext4 to NTFS or FAT).

I don't think that implementing an automatism that checks for lower-case and upper-case version is the right way because what happens if the directory (e.g. for ext4) contains the files toto.FCStd and toto.fcstd or Toto.fcstd? Or what happens if on this system there are the paths /home/user/Projects/toto.fcstd and /home/user/projects/toto.fcstd?
In this case the automatism may load the wrong file and the user can hardly control it.

yorikvanhavre, 2023-10-04:
> Or what happens if on this system there are the paths /home/user/Projects/toto.fcstd and /home/user/projects/toto.fcstd?

In #10930 It is done like this: If there is a file written exactly like the stored path, then that file is loaded. Otherwise, look for the same file with case-insensitive extension

Zolko-123, 2023-10-04:
> > The faulty references to FCStd or fcstd in an App::Link will cause a real error when loading a file
> 
> But at what point does it happen? Is it that the link property of the App::Link changes its content or is it that the referenced file changes its name?

I think it's the latter, when saving/loading/restoring assemblies and their files across different OSs and filesystems. Think PDM and a project developed by several people on their own machines, having different OSs, saving and copying and renaming files where needed : that's one of the main selling-points of FreeCAD, therefore we'd rather have this behave correctly

> I don't think that implementing an automatism that checks for lower-case and upper-case version is the right way because what happens if the directory (e.g. for ext4) contains the files toto.FCStd and toto.fcstd or Toto.fcstd ?

I only proposed considering upper- and lower-case letters for the _fcstd_ extension

wwmayer, 2023-10-04:
> Think PDM and a project developed by several people on their own machines, having different OSs, saving and copying and renaming files where needed

And does this affect only the file extension or also the file name? If the App::Link references a file Toto.fcstd that will become toto.fcstd then this issue is still present.

Zolko-123, 2023-10-04:
To be honest, 1 (one) such case was reported in the assembly forum. It concerned only the extension.

adrianinsaval, 2023-10-04:
> And does this affect only the file extension or also the file name? If the App::Link references a file Toto.fcstd that will become toto.fcstd then this issue is still present.

file extension changing is far more likely to happen though, when you change back from backup extension to .fcstd

btw on windows the filesystem is case sensitive but the system isn't, to the OS (including FreeCAD) file.fcstd is the same as file.FCStd but it's not the same in the filesystem so you don't notice the problem until you move to a case sensitive OS

jmickelin, 2023-10-07:
> FWIW, Microsoft specifically says that an extension is "three or four characters": [...]

This is just user level documentation describing what a (de facto) file extension is in non-technical words, not a normative standard for what modern-day developers should do. In fact, their developer documents say the opposite: Avoid short extensions, to prevent collisions and make the extension more self-documenting: https://learn.microsoft.com/en-gb/windows/win32/shell/fa-best-practices#do-not-use-short-file-name-extensions For a more concrete example of what they mean, the sibling document page called [File Association Example](https://learn.microsoft.com/en-gb/windows/win32/shell/fa-sample-scenarios) describes a toy example with an imaginary file extension called `.opa-ltw-audio`. That said, both `.fcstd` and `.fcad` should be self-documenting enough in my opinion.

> I was unable to find any actual standard, everything seems to be just de facto: does XDG have anything to say on the matter?

As far as I can recall, both XDG and Windows reference the extensions defined in the [IANA MIME Database](https://www.iana.org/assignments/media-types/media-types.xhtml) entries (where they exist), so if there are any limits imposed, it would be from them. But their registration form refers to for the file extension(s) field [RFC 6838 Section 4.12](https://www.rfc-editor.org/rfc/rfc6838.html#section-4.12), which doesn't specify any limits.