FreeCAD/FreeCAD #10876: [Problem]

mwoehlke, 2023-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

STEP export loses colors for certain assemblages.

To reproduce:
1. Create a Part and change its color
2. Add an additive cube primitive with default settings
3. Create an array feature with the preceding (change offset to 10mm)
4. Add an additive cylinder primitive with default settings

You should end up with something vaguely resembling a simple pea whistle, although any shape that forms a valid body using the above combination of features should work. The exact shape doesn't matter; the problem seems to manifest when adding features to a LinearPattern. (I didn't check other pattern types.) Naturally, the model on which I discovered this is quite a bit more complex, but the above is sufficient to show the issue. (My real model had Pads and Pockets after the LinearPattern.)

Select the Cylinder and export to STEP; the color shows up correctly. Select the Body and export to STEP; the color is missing. Delete the Cylinder, select the Body, and export to STEP; the color shows up correctly.

I can work around the issue by selecting the last feature of the problematic body (and, because I'm picky that way, manually editing the `.stp` to fix the part name), but in a complex model, that's a bit annoying. :slightly_smiling_face:

### Full version info

```shell
OS: Fedora Linux 38 (Thirty Eight) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Unknown
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.65
  * kicadStepUpMod 10.19.4
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-01:
Please reproduce on latest stable (v0.21.1) and report back

mwoehlke, 2023-10-01:
@luzpaz, I seem unable to fetch from GitHub at this time? In any case, it is going to be significant effort for me to try to build from source.

It takes *maybe* three minutes to test (less time than I've already spent trying to fetch the source), given the above test case. Did you try it? Were you unable to reproduce in v0.21.x?

mwoehlke, 2023-10-01:
Anyway, neither 0.21.1 nor master are building for me. The error (same in both branches) is quite inscrutable:
```
In file included from freecad/src/CXX/Objects.hxx:42,
                 from freecad/src/Base/PyObjectBase.h:56,
                 from freecad/src/Base/BaseClass.cpp:31:
freecad/src/CXX/Python3/Objects.hxx:1336:21: error: expected unqualified-id before ‘const’
 1336 |         SeqBase<T>( const Object &ob )
      |                     ^~~~~
```

luzpaz, 2023-10-03:
@Roy-043 do you mind testing this. I'm currently mobile and can't.

mwoehlke, 2023-10-06:
Right, and it eventually dawns on me that `SeqBase<T>( const Object &ob )` looks dodgy. At least, GCC ≥ 11 doesn't like it with `--std=c++20`. I'll try to take another look now that I'm past that build failure...

wwmayer, 2023-10-06:
There is an issue with gcc when trying to build as C++20. You have to choices to handle it:
* build with C++17 or
* use clang compiler

wwmayer, 2023-10-06:
> Select the Cylinder and export to STEP; the color shows up correctly. Select the Body and export to STEP; the color is missing. Delete the Cylinder, select the Body, and export to STEP; the color shows up correctly.

I can 100% confirm your observations.

mwoehlke, 2023-10-06:
> There is an issue with gcc when trying to build as C++20.

If by "issue" you mean GCC is obeying the standard. Filed as #10952.

> It has nothing to do with linear pattern. To me it looks like if the body has more than two objects the export has problems.

Could be; that just sounds like I didn't reduce my test case as much as was possible. :slightly_smiling_face:

Thanks for looking into this!

luzpaz, 2025-01-26:
> It has nothing to do with linear pattern. If instead of a pattern you create a second cube and attach it to any of the faces of the first cube and then create a cylinder or third cube the STEP export loses the colors. To me it looks like if the body has more than two objects the export has problems.

Do we need to improve the export logic ?

luzpaz, 2025-12-16:
@mwoehlke ping.. please retest on 1.1RC1 https://github.com/FreeCAD/FreeCAD/releases/tag/1.1rc1

ElementW, 2026-09-14:
Could reproduce in 0.20, 0.21, 1.0.0, and 1.1.3, but not `FreeCAD_weekly-2025.05.26-Linux-x86_64-py311.appimage`, the oldest weekly I have on hand at the moment, or any weeklies after that.

Clearly this existed at the 1.0.0 branch point but was fixed by something between it (2024-11-18) and the 2025.05.26 weekly.

ipatch, 2026-09-14:
could not reproduce using the file linked above by @FEA-eng 

used a combination of legacy and non legacy step export options, and opening / importing the `.step` files the color stays intact.

luzpaz, 2026-09-15:
Please add full About info for builds you test against. It saves time and provides a lot of clarity for posterity.

ElementW, 2026-09-15:
Well it's something among the many commits of [1.0.0...weekly-2025.04.21](https://github.com/FreeCAD/FreeCAD/compare/1.0.0...weekly-2025.04.21) (not d876e188082c314299f338799512965f3e25c03b because it's been backported to 1.0).

@luzpaz For versions, FC 1.1.3 is affected but not `weekly-2025.05.26`, yet:
* both use Qt 6.8.3
* both use Coin 4.0.3
* both use OCC 7.8.1

And I'm not going to drop 15+ full version dumps I've tested am I? (I mean, too late now, I'm not doing it after the fact)

luzpaz, 2026-09-15:
Thanks. that's fine, IMHO. No, that wouldn't make sense and thank you for all your work.

Roy-043, 2026-09-20:
Let's close then.

closed 2026-09-20 by Roy-043