FreeCAD/FreeCAD #10880: [Problem] TechDraw WB isn't showing dimensions.

mfraser, 2023-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Loaded a file this morning that I'd drawn in 0.21 into 0.22~pre1+git20230930.27053.fe4f1a4bc8-0ppa1~ubuntu22.04.1 and all of the dimensions are missing. Launching FreeCAD from the command line shows this:
```
6e-08 <App> Document.cpp(3135): exception in Tug_Boat#Dimension thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.000123935 <App> Document.cpp(3135): exception in Tug_Boat#Dimension006 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.000219676 <App> Document.cpp(3135): exception in Tug_Boat#Dimension007 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.0983538 <App> Document.cpp(3135): exception in Tug_Boat#Dimension002 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.0984627 <App> Document.cpp(3135): exception in Tug_Boat#Dimension003 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.0985375 <App> Document.cpp(3135): exception in Tug_Boat#Dimension001 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.0986087 <App> Document.cpp(3135): exception in Tug_Boat#Dimension004 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.0986799 <App> Document.cpp(3135): exception in Tug_Boat#Dimension005 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.0987523 <App> Document.cpp(3135): exception in Tug_Boat#Dimension009 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.0988286 <App> Document.cpp(3135): exception in Tug_Boat#Dimension013 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.273108 <App> Document.cpp(3135): exception in Tug_Boat#Dimension010 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.273224 <App> Document.cpp(3135): exception in Tug_Boat#Dimension011 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
0.273321 <App> Document.cpp(3135): exception in Tug_Boat#Dimension012 thrown: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
```
[Tug Boat.zip](https://github.com/FreeCAD/FreeCAD/files/12777218/Tug.Boat.zip)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.
Build type: Release
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.3.0
  * freecad.gears 1.0.0
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * fcgear 1.0.0
  * Lithophane
  * Freecad-Built-in-themes-beta 1.2.2
  * Curves 0.6.13
  * Assembly4 0.12.4
  * kicadStepUpMod 10.19.5
  * CurvedShapes 1.0.4
  * Defeaturing 1.2.0
  * DynamicData 2.53.0
  * fasteners 0.4.66
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.89.0
  * VendorParts
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-10-01:
The auto-correct logic is broken.  If you go to the Advanced tab in preferences and turn off "Auto Correct Dimension Refs" you'll be able to continue working while I fix it.

WandererFan, 2023-10-02:
should be resolved by PR #10880

luzpaz, 2023-10-07:
Closing. If problem persists, please re-open.

closed 2023-10-07 by luzpaz