FreeCAD/FreeCAD #10882: [Problem] Segfault importing DXF file

jes, 2023-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a particular DXF file that I'm working on in LibreCAD. I read that if you import a DXF file into FreeCAD then you can use the Path workbench to do the CAM, so I wanted to try it out.

When I tried to import the DXF file, FreeCAD segfaulted. Github won't let me attach the raw DXF file so I have gzipped it and attached that.
[scale.dxf.gz](https://github.com/FreeCAD/FreeCAD/files/12777493/scale.dxf.gz)

 It is supposed to look like this: [image]

FreeCAD says something about ignoring a degenerate arc of circle, may be related.

Here is the full text from the terminal output:

```$ ./FreeCAD_0.21.0-Linux-x86_64.AppImage 
FreeCAD 0.21.0, Libs: 0.21.0R33675 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
ImpExpDxf - ignore degenerate arc of circle
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f3ff9ca0090]
#1  /lib/x86_64-linux-gnu/libc.so.6(cfree+0x52) [0x7f3ff9cf7722]
#2  0x7f3f3c13828a in CDxfRead::ReadText() from /tmp/.mount_FreeCA4DjHeT/usr/lib/Import.so+0x87a
#3  0x7f3f3c13ca3c in CDxfRead::DoRead(bool) from /tmp/.mount_FreeCA4DjHeT/usr/lib/Import.so+0x53c
#4  /tmp/.mount_FreeCA4DjHeT/usr/lib/Import.so(+0x48bda) [0x7f3f3c0ebbda]
#5  /tmp/.mount_FreeCA4DjHeT/usr/lib/Import.so(+0x4048a) [0x7f3f3c0e348a]
#6  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADBase.so(method_varargs_call_handler+0x187) [0x7f3ffaf57417]
#7  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(+0x148c18) [0x7f3ffb81fc18]
#8  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x27b) [0x7f3ffb818f0b]
#9  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x5453) [0x7f3ffb814dc3]
#10  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x75) [0x7f3ffb820085]
#11  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b98) [0x7f3ffb814508]
#12  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(+0x1de9b0) [0x7f3ffb8b59b0]
#13  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(PyEval_EvalCode+0x8c) [0x7f3ffb8b58fc]
#14  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(+0x20db43) [0x7f3ffb8e4b43]
#15  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(+0x208efa) [0x7f3ffb8dfefa]
#16  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libpython3.10.so.1.0(PyRun_StringFlags+0x7d) [0x7f3ffb8d825d]
#17  0x7f3ffafbdeec in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADBase.so+0x6c
#18  0x7f3ffbfa1024 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x94
#19  0x7f3ffbfa11b3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0xe3
#20  0x7f3ffbea1a86 in Gui::Application::importFrom(char const*, char const*, char const*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x186
#21  0x7f3ffbfbb59c in StdCmdImport::activated(int) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x50c
#22  0x7f3ffbfa9f98 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x2a8
#23  0x7f3ffbfaa354 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x134
#24  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Core.so.5(+0x1dff9b) [0x7f3ffa27df9b]
#25  0x7f3ffaaaf136 in QAction::triggered(bool) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5+0x32
#26  0x7f3ffaab0c20 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5+0xc6
#27  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5(+0x259c3f) [0x7f3ffaba3c3f]
#28  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5(+0x25eb0e) [0x7f3ffaba8b0e]
#29  0x7f3ffaade202 in QWidget::event(QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5+0x134
#30  0x7f3ffaab4087 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#31  0x7f3ffaab836d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5+0x663
#32  0x7f3ffbf5e339 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x99
#33  0x7f3ffa260a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Core.so.5+0x8e
#34  0x7f3ffaab7981 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5+0x283
#35  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5(+0x1a2e02) [0x7f3ffaaece02]
#36  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5(+0x1a4505) [0x7f3ffaaee505]
#37  0x7f3ffaab4087 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#38  0x7f3ffbf5e339 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x99
#39  0x7f3ffa260a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Core.so.5+0x8e
#40  0x7f3ffa62e56e in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Gui.so.5+0x6b6
#41  0x7f3ffa61b091 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Gui.so.5+0x57
#42  /tmp/.mount_FreeCA4DjHeT/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5672c) [0x7f3ff493072c]
#43  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x2ad) [0x7f3ff665838d]
#44  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/./libglib-2.0.so.0(+0x5d649) [0x7f3ff6658649]
#45  /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f3ff66586de]
#46  0x7f3ffa291d50 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Core.so.5+0x5c
#47  0x7f3ffa25e5d8 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Core.so.5+0x180
#48  0x7f3ffa2627af in QCoreApplication::exec() from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libQt5Core.so.5+0x107
#49  0x7f3ffbec4bfe in Gui::Application::runApplication() from /tmp/.mount_FreeCA4DjHeT/usr/bin/../lib/libFreeCADGui.so+0x1e3e
#50  /tmp/.mount_FreeCA4DjHeT/usr/bin/freecad(+0x8608) [0x555a948c6608]
#51  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f3ff9c81083]
#52  /tmp/.mount_FreeCA4DjHeT/usr/bin/freecad(+0x8969) [0x555a948c6969]```

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (Regolith:GNOME-Flashback:GNOME/regolith)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.4.2
  * Render 2022.4.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jes, 2023-10-01:
I deleted the engraving marks, so that only the red parts remain, and now it loads correctly, so obviously the issue is something to do with the engraving.

luzpaz, 2023-10-02:
I can't reproduce segfault (on 0.22dev AppImage) but the DXF doesn't import correctly as seen from screenshot:

[image]

Report view shows:
`05:00:15  ImpExpDxf - ignore degenerate arc of circle`

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git) AppImage
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

yorikvanhavre, 2023-10-02:
The built-in importer does import the file but I see text rotations are not handled:
[image]
But the legacy importer works fine for me:
[image]

luzpaz, 2023-10-02:
@yorikvanhavre about info ?

wwmayer, 2023-10-06:
The crash was fixed two months ago: https://github.com/FreeCAD/FreeCAD/commit/15e9b45691e93cfda046f88e61bb6e812b628ba8

luzpaz, 2023-10-06:
Sorry, i was testing on an older v0.22dev appimage. I thought i downloaded one of the latest, that's weird

> The built-in importer does import the file but I see text rotations are not handled

@yorikvanhavre @wwmayer should we open a ticket for this part of the bug then ?

yorikvanhavre, 2023-10-09:
Maybe let's just change the ticket title, if that's ok for @jes

luzpaz, 2023-10-11:
@yorikvanhavre any suggestion who we could ask to assign this ticket too?

yorikvanhavre, 2023-10-11:
I'll look at it when I have a moment, it should be a simple fix

commit 55292e90 (2023-10-11): Import: Support DXF text rotation (#11001)

* Import: Support DXF text rotation - fixes #10882

- Reads and supports text rotation in builtin DXF import
- Makes the builtin DXF import produce Draft texts instead of App::Annotations
- Extends the arguments of Draft make_text()

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

---------
closed 2023-10-11 by Roy-043