FreeCAD/FreeCAD #10886: [Problem] Preferences dialog is space-inefficient and inflexible

kadet1090, 2023-10-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Navigation in the Preferences dialog is currently done via ListBox with huge icons which takes a lot of space making it hard to navigate around. Users also have to navigate using two styles of navigation - first user needs to pick category from the listbox on the left, and then pick proper tab from the main content window. It is space inefficient and creates a lot of unnecessary scrollable areas and truncated text.

[image]

### Full version info

```shell
0.22 dev
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2023-10-01:
I've created prototype on Figma with how this could work in FreeCAD: [https://www.figma.com/community/file/12 ... 1131468166](https://www.figma.com/community/file/1289973521131468166)

[image]

maxwxyz, 2024-03-09:
@kadet1090 could I close this issue with your implementation?

maxwxyz, 2024-04-21:
I'll close this as your PR was merged.

closed 2024-04-21 by maxwxyz